TheThingsNetwork/lorawan-stack #757: Application Server PubSub integration documentation

adriansmares, 2019-05-24:
#### Summary

The PubSub integration in the Application Server is implemented using the [gocloud.dev/pubsub](https://godoc.org/gocloud.dev/pubsub) package. However, there is no documentation available about how to use it (what is the format of the publish/subscribe URLs; are there any environment variables available required ?)

#### Why do we need this?

In order to make the user experience better when using the integration.

#### What is already there? What do you see now?

There is no documentation provided for now. [The original PR](https://github.com/TheThingsNetwork/lorawan-stack/pull/673#issue-278066318) describes the steps required to use the PubSub integration with an external NATS server.

#### What is missing? What do you want to see?

A separate document in `docs` which should describe how the integration should be used.

#### Environment

[AWS SQS](https://godoc.org/gocloud.dev/pubsub/awssnssqs), [GCP PubSub](https://godoc.org/gocloud.dev/pubsub/gcppubsub), [Azure SB](https://godoc.org/gocloud.dev/pubsub/azuresb) and [NATS](https://godoc.org/gocloud.dev/pubsub/natspubsub) are a good start.

#### How do you propose to implement this?

The documentation should describe at least what format should be used for the PublisherURLs and SubscriberURLs based on the driver.

#### Can you do this yourself and submit a Pull Request?

I can create this issue and review the results.

htdvisser, 2019-06-25:
Blocked on #401

adriansmares, 2020-05-01:
Superseeded by #2454. This issue was stale.

closed 2020-05-01 by adriansmares