TheThingsNetwork/lorawan-stack #781: Restrict password length

johanstokking, 2019-05-31:
#### Summary

Restrict password length.

#### Why do we need this?

For the server to not perform cryptographic operations on very large blocks of data.

#### What is already there? What do you see now?

Minimum password requirements.

#### What is missing? What do you want to see?

A maximum requirement on length of, say, 100 chars.

#### How do you propose to implement this?

Add a field to the [IS config](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/identityserver/identityserver.go#L54) and enforce it like the others.

#### Can you do this yourself and submit a Pull Request?

@furtiman please implement, @htdvisser is code owner for review

KrishnaIyer, 2019-05-31:
100 is a lot for a password. Most of the time even generated passwords are 16/32 chars. 
40 is good enough I think.

furtiman, 2019-05-31:
I agree with @KrishnaIyer

johanstokking, 2019-05-31:
Doesn't matter really, we store them [fixed length](https://en.wikipedia.org/wiki/PBKDF2) anyway. The goal is to reduce computational overhead. Also LastPass generated passwords go up to 100.

htdvisser, 2019-06-11:
I don't think there is any reason to limit password lengths to anything less than 100 (or even 1000) characters for performance reasons. Since we use PBKDF2-HMAC-SHA256 the computation time mostly depends on the number of iterations, and not really on the length of the plaintext password.

PR #784 (2019-06-11): IS maximum password length restriction

#### Summary

Closes #781 

#### Changes

- Added MaxLen attribute to password requirements
- Added default max length configuration of 100 symbols in cmd/.../identityserver/config.go
- Added error messages for min/max length violation
- Added tests
closed 2019-06-11 by furtiman