TheThingsNetwork/lorawan-stack #782: Research function in the doc

Sypheos, 2019-05-31:
#### Summary
Search function is common and essential function for any documentation.
...

#### Why do we need this?
The documentation must be searchable to help finding document related to a topic and find problem solving solution faster.

...

#### What is already there? What do you see now?
Nothing
...

#### What is missing? What do you want to see?
A search bar proposing result based on the input
...

#### Environment
Hugo, HTML, JS
...

#### How do you propose to implement this?
Hugo propose several option on there website.
https://gohugo.io/tools/search/
Algolia looks like a good choice for the doc hosted on github, they a free Community tiers that can fit.
A local search for offline documentation would be interesting too.
...

#### Can you do this yourself and submit a Pull Request?
Yes
...

Sypheos, 2019-05-31:
Blocked by #395

benolayinka, 2020-06-23:
Closed by #2588

PR #2588 (2020-06-22): Add search to docs

#### Summary

Closes #782 

#### Changes

- Add search to docs
- Add gh-pages config so `mage docs:build` uses separate envs

[image]
[image]

#### Testing

Run locally

##### Regressions

...

#### Notes for Reviewers

Beware when testing - the results return links to the thethingsstack.io, so clicking on a result will take you out of your local environment. This may also be an issue in the future
closed 2020-06-23 by benolayinka