TheThingsNetwork/lorawan-stack #783: Component documentation

Sypheos, 2019-05-31:
**Umbrella issue**

#### Summary
Services and components need a high level documentation to enable knowledge transfert and make it easier to maintain, improve components.
...

#### Why do we need this?
Without documentation maintaining and collaboration is nearly impossible. If we want to enable the community to contribute to and use the stack we need an high level documentation about each component and service about the stack. 

...

#### What is already there? What do you see now?
#395 All documentation is under `/docs/content`

...

#### What is missing? What do you want to see?
A `/docs/content/components` with high level description on all the services and components in the stack following this plan.

1. description / For what purpose
2. It's place in the architecture and/or role
3. What does it do.
4. Link annex resources (like API reference)
...

#### Environment
Hugo

...

#### How do you propose to implement this?
Each Codeowner write the documentation for the component he his the owner of.
Under `/docs/content/components/<component>.md` 
...

#### Can you do this yourself and submit a Pull Request?
Security
/pkg/auth @johanstokking @htdvisser
/pkg/crypto @johanstokking @htdvisser

Shared components
/pkg/component @johanstokking @htdvisser
/pkg/types @johanstokking @htdvisser @rvolosatovs
/pkg/webui @bafonins @kschiffer 

Subsystems
/pkg/applicationserver @johanstokking
/pkg/console @bafonins @kschiffer
/pkg/gatewayserver @johanstokking
/pkg/networkserver @johanstokking @rvolosatovs
/pkg/identityserver @htdvisser
/pkg/joinserver @johanstokking @rvolosatovs
...

htdvisser, 2019-06-11:
I think this kind of (code) documentation belongs in godoc, and not in Hugo. We could consider pulling package docs/comments from go files into markdown, but IMO the godoc should be leading for documenting our go packages.

Sypheos, 2019-06-11:
Not aiming to make a technical doc but more of an introduction, description of the components, how they work with the other services and what they provide with some example.

Maybe shorten the scope to the subsystems.

johanstokking, 2019-06-11:
There's a difference between component documentation and code documentation. This issue is about component documentation, see also the issue title and points 1-2.

Code documentation should be godoc, which addresses a different audience, may have code examples and usage guidelines, etc. Since the stack is not intended to be used as a library, we don't have to spend time on that now.

I do think, however, that the list of packages is confusing; there shouldn't be anything like that in our documentation. But I guess this is simply taken from `CODEOWNERS` to figure out who's most knowledgeable about the topic.

htdvisser, 2019-06-11:
Ok yeah, the list of packages (and especially the quoted part below) confused me.

> Security
> /pkg/auth @johanstokking @htdvisser
> /pkg/crypto @johanstokking @htdvisser
> 
> Shared components
> /pkg/component @johanstokking @htdvisser
> /pkg/types @johanstokking @htdvisser @rvolosatovs

Can we agree on the following list of tasks then?

- [ ] `/docs/content/concepts/security` @johanstokking @htdvisser (both lorawan and ttn-lw-stack)
- [ ] `/docs/content/components/_index` @johanstokking @htdvisser @bafonins @kschiffer (backend core component and base webui structure)
- [ ] `/docs/content/components/applicationserver` @johanstokking
- [ ] `/docs/content/components/console` @bafonins @kschiffer
- [ ] `/docs/content/components/gatewayserver` @johanstokking
- [ ] `/docs/content/components/gatewayconfigurationserver` @adriansmares
- [ ] `/docs/content/components/networkserver` @johanstokking @rvolosatovs
- [ ] `/docs/content/components/identityserver` @htdvisser
- [ ] `/docs/content/components/joinserver` @johanstokking @rvolosatovs

johanstokking, 2019-06-11:
LGTM

htdvisser, 2019-07-23:
No more discussion, @johanstokking is working on the structure.

htdvisser, 2019-11-12:
- [ ] Components / Network Server (@roman)
- [ ] Components / Application Server (@johan)
- [ ] Components / Join Server (@roman)
- [ ] Components / Console (@kschiffer) also need to fix the screenshot
- [ ] Components / Crypto Server (@johan)
- [ ] Components / Device Claiming Server (@johan)
- [ ] Components / Gateway Configuration Server (@KrishnaIyer)
- [ ] Reference / API (@htdvisser)
- [ ] Reference / Configuration / Gateway Server Options (@iamBatman)
- [ ] Reference / Configuration / Application Server Options (@johan)
- [ ] Reference / Configuration / Console Options (@kschiffer)
- [ ] Reference / Configuration / Gateway Configuration Server Options (@KrishnaIyer)

closed 2020-02-27 by kschiffer