TheThingsNetwork/lorawan-stack #789: Duplicate events

rvolosatovs, 2019-06-03:
#### Summary

NS publishes an event whenever a MAC command is added to a buffer(in `generateDownlink`), however that does not equate to NS actually scheduling the MAC command. Additionally, the downlink may be generated twice (e.g. in case Rx1 is not possible for Class C device)

#### Steps to Reproduce

1. Look at the source

#### What do you see now?

Duplicate/inapplicable events

#### What do you want to see instead?

Non-duplicate, applicable events

#### How do you propose to implement this?

Return a slice of events from `generateDownlink`

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2019-06-05:
Note that the issue also occurs when processing the uplink - in case the transaction fails and we retry(#33), the events would be duplicated

htdvisser, 2019-06-14:
> **How do you propose to implement this?**
> Return a slice of events from `generateDownlink`

This description of a proposed implementation needs a bit more information before I can decide whether I agree with the proposal or not. 

Maybe this was discussed with @johanstokking offline while I was on vacation, but I would still like to see some more details in this issue.

johanstokking, 2019-06-14:
FTR this wasn't discussed offline

rvolosatovs, 2019-06-14:
That's very simple really, MAC handlers/enqueuers shall not publish events, but rather return events, which NS can later decide to publish.

rvolosatovs, 2019-06-14:
https://github.com/rvolosatovs/lorawan-stack/blob/c10ca5fed8e317926b60cfb71e9f70c2414b3419/pkg/networkserver/downlink.go#L104

And so if NS downlink was successfully processed, NS does this:(`queuedEvents` contains the `Events` returned `generateDownlink` at that point)
https://github.com/rvolosatovs/lorawan-stack/blob/c10ca5fed8e317926b60cfb71e9f70c2414b3419/pkg/networkserver/downlink.go#L1198-L1204

PR #1391 (2019-09-29): Only publish MAC downlink events on success

#### Summary

Closes #789 

#### Changes

- Only publish MAC downlink events on successful scheduling
- Fix invalid CIDs used for `RejoinParamSetupReq` and `PingSlotReq` MAC commands
- Wrap multiple return values of MAC enqueuers into a struct

#### Notes for Reviewers

Tested on metal
closed 2019-09-29 by rvolosatovs