TheThingsNetwork/lorawan-stack #790: Update console dependencies

bafonins, 2019-06-04:
#### Summary

We should update react and react related dependencies.

#### Why do we need this?

Keep up with the latest versions of dependencies. What is more, the [16.6](https://github.com/facebook/react/releases/tag/v16.6.0) version adds several useful features, namely `React.memo()` and `contextType`.

#### What is already there? What do you see now?

Old versions

#### What is missing? What do you want to see?

Updated dependencies
deps:
- [ ] react
- [ ] react-dom
- [ ] redux
- [ ] redux-logic
- [ ] connected-react-router
- [ ] react-redux, no need to update now, but might be useful later
- [ ] formik, v2 of the library introduces variety of cool features, but is still in development
- [ ] react-intl, v3 is in beta (no major changes though)
 
dev-deps:
- [ ] storybook, there v5 out and looks very nice, however i can imagine that we would have to change our storybook config

#### Environment

browser

#### How do you propose to implement this?

Update deps in `package.json` and make sure nothing breaks

#### Can you do this yourself and submit a Pull Request?

yes. would be nice to update major libraries before the console mvp

cc @kschiffer fill free to add/update stuff

kschiffer, 2019-06-04:
I'm actually partially working on this atm via #772, at least as far as some devDeps (including storybook) go. Should see a PR soon.

kschiffer, 2019-06-06:
Description updated

bafonins, 2019-08-15:
Description updated

closed 2020-06-16 by bafonins