TheThingsNetwork/lorawan-stack #794: Upgrade TxSettings.timestamp to int64

KrishnaIyer, 2019-06-05:
#### Summary

Upgrade the TxSettings.timestamp from `uint32` to `int64`.

#### Why do we need this?

Some gateways (ex: TTIG using basic station) use `int64` for the concentrator timestamp. 
 Theoretically, the timestamp cannot cross 32 bits until 2038 but some gateways may not be (ntp) time-synced and this means that they might have a concentrator timestamp that exceeds `uint32`.

#### What is already there? What do you see now?

Timestamp is `uint32`.

#### What is missing? What do you want to see?

Timestamp is made `int64` (`uint64` if necessary).

#### Environment

TTIG with Basic Station.

#### How do you propose to implement this?

- Update `lorawan.proto` and regenerate.
- Fix all occurrences in the code  (mostly GS, NS and `ttnpb/udp`).
- Verify that tests pass.

#### Can you do this yourself and submit a Pull Request?

Yes. But if @furtiman wants, he can take this as well.

cc: @johanstokking @rvolosatovs

furtiman, 2019-06-05:
Will do 👍

johanstokking, 2019-06-05:
The SX1301 timestamp is `uint32` in microseconds. If BS uses `int64`, the overflow should be handled out of band, but I'm quite sure it's all concentrator timestamp in the end.

closed 2019-06-05 by johanstokking

KrishnaIyer, 2019-06-06:
@johanstokking: What you say is right but Basic Station needs the Upper 32 bits of the timestamp to synchronize downlink. There are 2 ways of doing this:
a. Upgrade the timestamp and use only the lower 32 bits for all existing operations.
b.  Maintain the upper byte per gateway in the GS frontend.

I prefer the option `a` since it's the easiest to implement.

johanstokking, 2019-06-11:
The `timestamp` is strictly SX130X timestamp and that's `uint32`. Also we rely on rollovers because it's a `uint32` so we break things when we go to `int64`.

What are the MSB used for exactly? Because there's still the same SX1308 concentrator in there which uses `uint32`, so I guess this is a BS specific thing that we take care of in the BS frontend.

Is it worth considering making this part of the downlink tokens if it's related to downlink? We already preserve state there, albeit shortly.

KrishnaIyer, 2019-06-11:
Ah yes indeed. downlink tokens is the best place for this. 
The MSB used as a "session identifier" which needs to be echoed to the station on downlinks.