TheThingsNetwork/lorawan-stack #805: Can not serve the frontend for development

twowinter, 2019-06-11:
#### Summary

I was successfully build the frontend, and saw the console login page on http://localhost:1885 . Then i tried to serve the frontend following the guide, but the website http://localhost:8080 showed "Cannot GET /".

#### Steps to Reproduce

1. make init
2. ./mage init
3. ./mage js:build
4. go build ./cmd/ttn-lw-stack/
5. ./cmd/ttn-lw-stack start all
6. In browser -> http://localhost:1885/console
7. setup environment

```bash
export NODE_ENV=development
export TTN_LW_LOG_LEVEL=debug
export TTN_LW_IS_OAUTH_UI_JS_FILE=libs.bundle.js,oauth.js
export TTN_LW_CONSOLE_UI_JS_FILE=libs.bundle.js,console.js
export TTN_LW_CONSOLE_UI_CANONICAL_URL=http://localhost:8080/console
export TTN_LW_CONSOLE_OAUTH_AUTHORIZE_URL=http://localhost:8080/oauth/authorize
export TTN_LW_CONSOLE_OAUTH_TOKEN_URL=http://localhost:8080/oauth/token
export TTN_LW_IS_OAUTH_UI_CANONICAL_URL=http://localhost:8080/oauth
export TTN_LW_IS_EMAIL_NETWORK_IDENTITY_SERVER_URL=http://localhost:8080/oauth.js
export TTN_LW_CONSOLE_UI_ASSETS_BASE_URL=http://localhost:8080/assets
```

8. ./mage js:serve
9. In browser -> http://localhost:8080

#### What do you see now?

[image]

```bash
$ ./mage js:serve
ℹ ｢wds｣: Project is running at http://localhost:8080/
ℹ ｢wds｣: webpack output is served from /assets
ℹ ｢wds｣: 404s will fallback to /index.html
Executing additional scripts before exit
ℹ ｢wdm｣:    1958 modules
ℹ ｢wdm｣: Compiled successfully.
```

#### What do you want to see instead?

...

#### Environment

Node v10.15.0
NPM 6.9.0
ttn-lw-stack version v3.0.3

#### How do you propose to implement this?

I think it may be that I am missing any steps, and these steps may be described in detail in the documentation.

#### Can you do this yourself and submit a Pull Request?

...

bafonins, 2019-06-11:
Currently, the console in dev mode is served from `localhost:8080/console`. Could you try navigating to `/console` instead of just `/`?

twowinter, 2019-06-11:
> Currently, the console in dev mode is served from `localhost:8080/console`. Could you try navigating to `/console` instead of just `/`?

nothing showed. here is the log.

[image]

bafonins, 2019-06-11:
After quick debugging it turned out that the problem is in these lines:
```
export TTN_LW_IS_OAUTH_UI_JS_FILE=libs.bundle.js,oauth.js
export TTN_LW_CONSOLE_UI_JS_FILE=libs.bundle.js,console.js
```

the list of javascript files to be served to the client is not properly defined which causes the browser to request for `/assets/libs.bundle.js,console.js` instead of `/assets/libs.bundle.js` and `/assets/console.js`

Could you try changing the those to
```
export TTN_LW_IS_OAUTH_UI_JS_FILE="libs.bundle.js oauth.js"
export TTN_LW_CONSOLE_UI_JS_FILE="libs.bundle.js console.js"
```
and run the stack again?

Note, that since you deploy the `oauth` app on `8080` you have to change the oauth `redirect-uri` address in [dev.make L44](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/.make/dev.make#L44) from `http://localhost:1885/console/oauth/callback` to `http://localhost:8080/console/oauth/callback` and re-register the console again (`make dev.stack.init`). Will add this to the documentation.

twowinter, 2019-06-11:
Hi @bafonins , it seem the same on my side, there is no response for the 200 request.

[image]

bafonins, 2019-06-11:
1. Could you check the console in the browser dev tools for any warnings/errors?
2. Could you verify that you have `NODE_ENV` set to `development`?  There should be no hashes appended to the frontend assets when in the development mode. I also dont see `libs.bundle.js` being recognised and served by `webpack-dev-server`.

Here is the output from `webpack-dev-server` I have:
```
$ mage js:clean js:build js:serve
[HPM] Proxy created: [ '/console', '/oauth', '/api' ]  ->  http://localhost:1885
ℹ ｢wds｣: Project is running at http://localhost:8080/
ℹ ｢wds｣: webpack output is served from /assets/
ℹ ｢wds｣: Content not from webpack is served from .../go.thethings.network/lorawan-stack
ℹ ｢wds｣: 404s will fallback to /index.html
ℹ ｢wdm｣: wait until bundle finished: /assets/console.css
ℹ ｢wdm｣: wait until bundle finished: /assets/libs.bundle.js
ℹ ｢wdm｣: wait until bundle finished: /assets/console.js
```

twowinter, 2019-06-12:
Hi, @bafonins . Reason was caught. 

The request url is http://localhost:8080/assets/console.a99f5eca0b9c3ce9a0be.js, with hash appended to the frontend assets. 

[image]

I'm not sure if this URL comes from /assets/index.html, it might be affected by stack. After I rebuild the stack under the 'NODE_ENV=development' environment, there is no longer hashes appended to the request assets. And i can see the page now.

So, according to this issue, i think the development guide may need to be a bit clearer.

```text
After following the Getting Started section to initialize the stack and doing an initial build of the frontend via mage js:build, it can be served using:

export NODE_ENV=development
mage js:serve
```

This description steps is incorrect, the environment need to be set before the build of frontend and stack.

bafonins, 2019-06-12:
> This description steps is incorrect, the environment need to be set before the build of frontend and stack.

[Build Process](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/DEVELOPMENT.md#build-process), but I can agree that the next paragraph can be misleading.

Will update the documentation before releasing the console.

closed 2019-06-12 by bafonins