TheThingsNetwork/lorawan-stack #808: Support Unmarshalling of SX1301 Conf

KrishnaIyer, 2019-06-11:
#### Summary

Support Unmarshalling of SX1301 Conf

#### Why do we need this?

To un-marshal SX1301 Configuration when used stand-alone or as part of packet-forwarder/router config.

#### What is already there? What do you see now?

Not Implemented `errUnmarshalNotImplemented `

#### What is missing? What do you want to see?

Implement it.

#### Environment

`ttn-lw-stack v3.03`

#### How do you propose to implement this?

- Multi-stage un-marshaling due to the sub-messages.

#### Can you do this yourself and submit a Pull Request?

Yup

KrishnaIyer, 2019-07-23:
Closing this since we don't really need this.

KrishnaIyer, 2019-08-01:
As discussed in the previous status meeting, I'm implementing a simple un-marshaler to support Basic Station tests. 
If we need to optimize it,

johanstokking, 2019-09-11:
Please consider a `map[string]interface{}` for testing and keep it simple. At runtime I don't see a reason to unmarshal the SX1301 configuration.

KrishnaIyer, 2019-09-11:
Yeah that's the plan. This is only for testing so I don't plan to spend time on optimizing it.

PR #1366 (2019-09-20): Support unmarshaling of SX1301 Conf Messages

#### Summary

Closes #808 

#### Changes

- Implement unmarshaling.
- Add corresponding test to Basic Station.

#### Notes for Reviewers

- As discussed in the issue, this is only to support tests.

#### Release Notes

- Added unmarshaling support to SX1301Conf.
closed 2019-09-20 by htdvisser