TheThingsNetwork/lorawan-stack #830: Use registry for AS pub/sub configuration per application

johanstokking, 2019-06-19:
#### Summary

Use registry for AS pub/sub configuration per application

#### Why do we need this?

To differentiate pub/sub configuration(s) per application, i.e. whether they're enabled, what endpoints, etc

#### What is already there? What do you see now?

Since #673 we have pub/sub configuration on the AS instance level. This met the requirements, but doesn't allow for enough flexibility at runtime and does not provide enough application context.

#### What is missing? What do you want to see?

Use a webhooks-like registry with configuration per application.

#### How do you propose to implement this?

- Add a configuration proto message for the pub/sub configuration per application (like `ApplicationWebhook`), i.e. `ApplicationPubSub`. @htdvisser should we store the `go-cloud` url (pro: easy, con: we probably need some config help in CLI and Console and gives lock-in to `go-cloud`) or a one-of per supported pub/sub from where we build the url?
- Add a registry and Redis implementation for that (only get/set/delete)
- Add a service and rpcs to manage the pub/sub config
- Wire the subscription just like webhooks

It is basically very much like webhooks is w.r.t. configuration and wiring. 

#### Can you do this yourself and submit a Pull Request?

Will review

htdvisser, 2019-06-19:
I think it would be smarter to have per-pub/sub fieldsets. I don't think that we'll be able to use the go-cloud urls in production (because some implementations will then rely on environment variables for auth). Instead, I think we'll instantiate the pub/sub clients explicitly and only use the benefits of their common interfaces.

adriansmares, 2019-06-20:
Since we are dropping the URL support (for good reasons), is there any point in keeping the implementation bound to `go-cloud` or should we go for the native libraries instead ?

johanstokking, 2019-06-20:
Let's make our own life easy and go with `go-cloud`

adriansmares, 2019-06-25:
I've been working on this for a while and currently the following progress has been made:
- `ApplicationPubSub` proto has been added. It is very similar to a `ApplicationWebHook`, but it has attributes and a specific service (NATS, AWS etc.) associated with it.
- `ApplicationPubSub` registry, with context support.
- gRPC support for List/Get/Set/Delete.
- Integration tasks - basically tasks that create the connection to the PubSub provider in the background when the integration is created / updated. Similar to how `ApplicationLink` linking works.
- NATS provider support - this time without any ugly environment variables. Server is specified by the `nats-server` attribute. 

I'll finish up the tests and submit a PR soon.

PR #883 (2019-07-05): Application PubSub registry

#### Summary

Closes #830

#### Changes

- PubSub integrations have been changed to work akin to how Webhooks work - they are stored inside a registry instead of being part of the application server configuration, and now exist on a per-application basis.
- Added the PubSub registry and gRPC support.
- Added CLI support for PubSub integrations.
- The NATS provider is no longer dependent on additional environment variables  - the client is manually created on a per-integration basis. This applies to other integrations which will be added in the future as well, since [... cut]
closed 2019-07-05 by adriansmares