TheThingsNetwork/lorawan-stack #833: Support Semtech Join Server

johanstokking, 2019-06-20:
#### Summary

Implement join flow through Semtech Join Server

#### Why do we need this?

Support Semtech modem and other devices provisioned on the Semtech Join Server

#### What is already there? What do you see now?

- Join flow through V3 Join Server (gRPC)
- Interop server and messages for Backend Interfaces
- Existing requirement for join server lookup and authorization (#4)

#### What is missing? What do you want to see?

Implement Semtech Join Server, see [documentation](https://tracknet.bitbucket.io/js-doc/)

This entails;
- Join flow
- Autoprovisioning of devices
- Out of scope for this issue; change of ownership

There's support for Backend Interfaces, although with custom authentication, lookup and paths.

There's also support for websockets, but I only see use for that if we keep a pool of open websocket connections that we reuse to avoid TLS handshake for each activation.

For portability, I would go for the Backend Interfaces approach since we can also use the messages in `pkg/interop`

#### How do you propose to implement this?

- Extend `pkg/interop` with a client for the join flow. It would be nice to have options for (custom) authentication, lookup and paths
- We probably need configuration of these options per JoinEUI prefix in NS. This is a level of out-of-band configuration that's definitely not ideal but since there's no standard imposed for authentication and lookup, we have to deal with this
- Incorporate this in Network Server, needs #4 

#### Can you do this yourself and submit a Pull Request?

Will review

htdvisser, 2019-06-21:
@johanstokking you marked this issue as blocked. Please add a comment to indicate on what it's blocked, so that we can unblock this issue if the blocking issue is resolved.

johanstokking, 2019-06-21:
It says "Needs #4"

htdvisser, 2019-06-21:
Ah. _edited_. My email didn't show that line. 🤦‍♂

rvolosatovs, 2019-08-08:
@johanstokking can you clarify
> Autoprovisioning of devices

What does it entail exactly?
Is #1004 enough to close this issue?

johanstokking, 2019-08-08:
I think they have a mechanism to provision devices on a NS/AS with some profile.

Let close this issue in #1004 

Please do check the spec and file an issue for it in Backlog if this makes any sense

PR #1004 (2019-08-09): Add client for JS interoperability

#### Summary

Closes #4 
Closes #833 (it's just setting a custom fqdn and path)

#### Changes

- Add `interop.Client`, which handles JS requests via DNS lookup
- Add `interop.LoRaAllianceJoinServerFQDN`, which can be used to generate the FQDN for the DNS lookup using LoRa Alliance DNS service
- Fallback to interop client in NS and AS if cluster-local JS peer is not found

#### Notes for Reviewers

After this is merged I will create a (goldplating+performance) issue to allow e.g. caching of the connections via `interop.Client`

#### Release Notes
closed 2019-08-09 by rvolosatovs