TheThingsNetwork/lorawan-stack #834: Flaky NS (Redis) test

KrishnaIyer, 2019-06-20:
#### Summary

There's one test that's consistently failing randomly 😛  

#### Steps to Reproduce

1. Push to a PR
2. Check travis output.
3. Either the `ARCH=amd64` or the `ARCH=386` fails randomly

#### What do you see now?

[image]

```
namespace=networkserver task_id=process_downlink test_name=TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2
--- FAIL: TestFlow (0.00s)
    --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue (32.78s)
        --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2 (32.78s)
            networkserver_flow_test.go:208: Cluster.GetPeer called
            networkserver_flow_test.go:208: ApplicationAccess.ListRights called
            networkserver_flow_test.go:214: Cluster.GetPeer called
            networkserver_flow_test.go:214: ApplicationAccess.ListRights called
            networkserver_flow_test.go:240: Device created
            networkserver_flow_test.go:322: Cluster.GetPeer called
            networkserver_util_test.go:458: Cluster.Auth called
            networkserver_flow_test.go:460: Cluster.GetPeer called
            networkserver_util_test.go:488: Cluster.Auth called
            networkserver_flow_test.go:508: Device successfully joined. DevAddr: C03E4049
            --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2/uplink (32.75s)
                networkserver_flow_test.go:573: Timed out while waiting for deduplication window to close
            networkserver_flow_test.go:562: HandleUplink returned rpc error: code = Unavailable desc = transport is closing
FAIL
FAIL	go.thethings.network/lorawan-stack/pkg/networkserver	39.564s
```

#### What do you want to see instead?

No failure as for the other build config.

#### Environment

Travis, `ttn-lw-stack: v3.0.3`

#### How do you propose to implement this?

Up to the NS expert @rvolosatovs 

#### Can you do this yourself and submit a Pull Request?

Nopes

johanstokking, 2019-06-20:
Got this as well

[error.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3312140/error.txt)

rvolosatovs, 2019-06-21:
Sometimes Travis is slow and the test times out, so the delay(https://github.com/TheThingsNetwork/lorawan-stack/blob/6eb8fe05751cd81b0dd16a97a05de67356d830d2/pkg/networkserver/networkserver_flow_test.go#L803) should be increased again

PR #844 (2019-06-26): Make JS/GS/NS/AS optional

#### Summary

Closes #834 (not related but super annoying and small "fix")
Closes #842 
Closes #843 

#### Changes

- Make JS, GS, NS and AS optional through config. This is like Console
- Set device in JS when any JS field is set (not just `root_keys`)

#### Release Notes

- Added CLI support for enabling/disabling JS, GS, NS and AS through configuration
closed 2019-06-26 by johanstokking