TheThingsNetwork/lorawan-stack #843: CLI requires NS and AS for creating devices in JS

johanstokking, 2019-06-20:
#### Summary

CLI requires NS and AS for creating devices in JS

#### Steps to Reproduce

1. Create a device with only JS fields, i.e. `$ ttn-lw-cli dev create test-app test-dev --join-eui 0102030405060708 --dev-eui 01020304050607ff --root-keys.app-key.key 01020304050607080102030405060708 --resets-join-nonces=true --net-id=000001 --network-server-address 000001`

#### What do you see now?

- Since `abp` is unset, `supports_join` is set, which is an NS field, triggering a NS hit for create/update
- On device create, [NS gets hit anyway](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices_split.go#L221)
- But NS requires a device profile to be set, i.e. LoRaWAN MAC and PHY versions

#### What do you want to see instead?

I want to be able to create devices in JS only

#### How do you propose to implement this?

Force CLI not to call JS/NS/AS. This is similar to the Console, where we disable functionality when the component isn't there.

#### Can you do this yourself and submit a Pull Request?

Will review

johanstokking, 2019-06-20:
Also, the CLI currently requires `lorawan_version` to be set in all circumstances, while it's only necessary when creating ABP devices for validating the session keys. Suggesting;

```diff
diff --git a/cmd/ttn-lw-cli/commands/end_devices.go b/cmd/ttn-lw-cli/commands/end_devices.go
index 2fd69e78..b20c9297 100644
--- a/cmd/ttn-lw-cli/commands/end_devices.go
+++ b/cmd/ttn-lw-cli/commands/end_devices.go
@@ -294,19 +294,6 @@ var (
 				paths = append(paths, ttnpb.FlattenPaths(decodedPaths, endDeviceFlattenPaths)...)
 			}
 
-			var macVersion ttnpb.MACVersion
-			s, err := setEndDeviceFlags.GetString("lorawan_version")
-			if err != nil {
-				return err
-			}
-
-			if err := macVersion.UnmarshalText([]byte(s)); err != nil {
-				return err
-			}
-			if err := macVersion.Validate(); err != nil {
-				return errInvalidMACVerson
-			}
-
 			setDefaults, _ := cmd.Flags().GetBool("defaults")
 			if setDefaults {
 				device.NetworkServerAddress = getHost(config.NetworkServerGRPCAddress)
@@ -344,6 +331,17 @@ var (
 						"session.keys.app_s_key.key",
 						"session.dev_addr",
 					)
+					var macVersion ttnpb.MACVersion
+					s, err := setEndDeviceFlags.GetString("lorawan_version")
+					if err != nil {
+						return err
+					}
+					if err := macVersion.UnmarshalText([]byte(s)); err != nil {
+						return err
+					}
+					if err := macVersion.Validate(); err != nil {
+						return errInvalidMACVerson
+					}
 					if macVersion.Compare(ttnpb.MAC_V1_1) >= 0 {
 						device.Session.SessionKeys.SNwkSIntKey = &ttnpb.KeyEnvelope{Key: generateKey()}
 						device.Session.SessionKeys.NwkSEncKey = &ttnpb.KeyEnvelope{Key: generateKey()}
```

cc @rvolosatovs

htdvisser, 2019-06-21:
I think it could be an idea to add (`--skip-is`,) `--skip-ns`, `--skip-as` and `--skip-js` flags to allow advanced users to skip creates/updates on the (IS/)NS/AS/JS. The `--skip-is` may be a bit tricky, I think we shouldn't allow that one for creates.

As far as I know the MAC version is very useful, also for OTAA devices. If it's not now, it could still be valuable to have it in the future, so I think it would be wise to keep it where it is.

johanstokking, 2019-06-21:
> I think it could be an idea to add (`--skip-is`,) `--skip-ns`, `--skip-as` and `--skip-js` flags to allow advanced users to skip creates/updates on the (IS/)NS/AS/JS. The `--skip-is` may be a bit tricky, I think we shouldn't allow that one for creates.

How about skipping them when the concerning `..._grpc_address` is empty?

> As far as I know the MAC version is very useful, also for OTAA devices. If it's not now, it could still be valuable to have it in the future, so I think it would be wise to keep it where it is.

It's not known when provisioning devices on the JS, for example when provisioning secure elements long before the end device's firmware is chosen. JS doesn't care about MAC version either, it works with the selected MAC version from NS in the join flow.

htdvisser, 2019-06-21:
I don't think it's a good idea to look at address config when deciding where to create end devices, it's just too easy to make mistakes with that (when forgetting to configure an env var). Skipping end device creation on specific server types is a special use case that deserves explicit flags, and should not implicitly be decided when certain config is omitted.

---

Removing the MAC version is IMO out of scope for this issue and should be discussed with @rvolosatovs before moving forward with that.

johanstokking, 2019-06-21:
> Skipping end device creation on specific server types is a special use case that deserves explicit flags, and should not implicitly be decided when certain config is omitted (possibly by accident).

Standalone AS and standalone JS are not special use cases. Let's go for bools then, like Console.

rvolosatovs, 2019-06-21:
Regarding the `lorawan_version` - I agree with @johanstokking, we only need it in CLI for creating ABP devices, so no reason to require it.
`bool` flags sound good!

PR #844 (2019-06-26): Make JS/GS/NS/AS optional

#### Summary

Closes #834 (not related but super annoying and small "fix")
Closes #842 
Closes #843 

#### Changes

- Make JS, GS, NS and AS optional through config. This is like Console
- Set device in JS when any JS field is set (not just `root_keys`)

#### Release Notes

- Added CLI support for enabling/disabling JS, GS, NS and AS through configuration
closed 2019-06-26 by johanstokking