TheThingsNetwork/lorawan-stack #845: webhook downlink no rights to application

ecities, 2019-06-24:
#### Summary

Problem when downlink message via webhooks:
**error:pkg/auth/rights:no_application_rights (no rights for application `app3`)**

...

#### Steps to Reproduce

1. ...Clone lorawan-stack from github (as 24.06.2019) + gettingstarted instruction
2. ... add app, dev, webhook, key as in gettingstarted
3. ... add all ACLs to application
4... webhook successfully send POST data
5.. Insufficient access rights to put data

root@propc:/home/robert# **ttn-lw-cli applications create app3 --user-id admin**

{
  "ids": {
    "application_id": "app3"
  },
  "created_at": "2019-06-24T12:21:16.862Z",
  "updated_at": "2019-06-24T12:21:16.862Z"
}

root@propc:/home/robert# **ttn-lw-cli end-devices create app3 dev4 --dev-eui 10217E00000003FF --app-eui A000000000000002 --frequency-plan-id EU_863_870 --root-keys.app-key.key 44D70CFA703C740D8EAB6A2811732B76 --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b**

Flag --app-eui has been deprecated, use the join-eui flag

{
  "ids": {
    "device_id": "dev4",
    "application_ids": {
      "application_id": "app3"
    },
    "dev_eui": "10217E00000003FF",
    "join_eui": "A000000000000002"
  },
  "created_at": "2019-06-24T12:21:34.430Z",
  "updated_at": "2019-06-24T12:21:34.667620964Z",
  "attributes": {
  },
  "network_server_address": "localhost",
  "application_server_address": "localhost",
  "join_server_address": "localhost",
  "lorawan_version": "1.0.2",
  "lorawan_phy_version": "1.0.2-b",
  "frequency_plan_id": "EU_863_870",
  "supports_join": true,
  "root_keys": {
    "app_key": {
      "key": "44D70CFA703C740D8EAB6A2811732B76"
    }
  }
}

root@propc:/home/robert# **ttn-lw-cli applications api-keys create --name link --application-id app3 --right-application-link --right-application-all --right-application-delete --right-application-devices-read --right-application-devices-read-keys --right-application-devices-write --right-application-devices-write-keys --right-application-info --right-application-link --right-application-settings-api-keys --right-application-settings-basic --right-application-settings-collaborators --right-application-traffic-down-write --right-application-traffic-read --right-application-traffic-up-write**

  INFO API key ID: 4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ
  INFO API key value: NNSXS.4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ.KNZJFT6R3DYLI5N5XZJFWCXDDYJBF2HAXHOCK2TXDBMTFCET3KFQ
  WARN The API key value will never be shown again
  WARN Make sure to copy it to a safe place
{
  "id": "4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ",
  "key": "NNSXS.4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ.KNZJFT6R3DYLI5N5XZJFWCXDDYJBF2HAXHOCK2TXDBMTFCET3KFQ",
  "name": "link",
  "rights": [
    "RIGHT_APPLICATION_DEVICES_WRITE",
    "RIGHT_APPLICATION_SETTINGS_BASIC",
    "RIGHT_APPLICATION_SETTINGS_COLLABORATORS",
[... 219 lines cut ...]

#### What do you see now?

root@propc:/home/robert# curl http://localhost:1885/api/v3/as/applications/app3/webhooks/wh4/devices/dev4/down/push -X POST   -H 'Authorization: Bearer "NNSXS.GIPVAVIAO77RFBOWTDO43XIZ5SDTRPDX7QKOGCQ.EJJRVXWEX7LIFSPYMT4WBPPSCIMIJEXTRMFOSRNT47RPF6KTWCJQ'  --data '{"downlinks":[{"frm_payload":"vu8=","f_port":15,"priority":"NORMAL"}]}'

error:pkg/auth/rights:no_application_rights (no rights for application `app3`)

...

#### What do you want to see instead?

...

#### Environment

UBUNTU 16.04LTS/FF/iFemtoCell+ SPF/TELIT RE866
...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2019-06-24:
Thanks @ecities, we'll have a look at this

johanstokking, 2019-06-25:
How did you create API key with value `NNSXS.GIPVAVIAO77RFBOWTDO43XIZ5SDTRPDX7QKOGCQ.EJJRVXWEX7LIFSPYMT4WBPPSCIMIJEXTRMFOSRNT47RPF6KTWCJQ` ?

Because the one you mention in your steps is different and that one has all the rights.

For sending downlink, you need `RIGHT_APPLICATION_TRAFFIC_DOWN_WRITE`.

ecities, 2019-06-25:
Hi Johan,
Thank you very much for fast replay.

**It looks I have this right listed.**
What I did:
I get --help info for command, and display all rights available and add all of it to the command:

 **ttn-lw-cli applications api-keys create --name link --application-id app3 --right-application-link --right-application-all --right-application-delete --right-application-devices-read --right-application-devices-read-keys --right-application-devices-write --right-application-devices-write-keys --right-application-info --right-application-link --right-application-settings-api-keys --right-application-settings-basic --right-application-settings-collaborators --right-application-traffic-down-write --right-application-traffic-read --right-application-traffic-up-write**

INFO API key ID: 4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ
INFO API key value: NNSXS.4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ.KNZJFT6R3DYLI5N5XZJFWCXDDYJBF2HAXHOCK2TXDBMTFCET3KFQ
WARN The API key value will never be shown again
WARN Make sure to copy it to a safe place
{
"id": "4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ",
"key": "NNSXS.4DU2SXFXICLJUGHJPAC6BUFYFHZSKT6SOBXABUQ.KNZJFT6R3DYLI5N5XZJFWCXDDYJBF2HAXHOCK2TXDBMTFCET3KFQ",
"name": "link",
"rights": [
"RIGHT_APPLICATION_DEVICES_WRITE",
"RIGHT_APPLICATION_SETTINGS_BASIC",
"RIGHT_APPLICATION_SETTINGS_COLLABORATORS",
"RIGHT_APPLICATION_TRAFFIC_UP_WRITE",
"RIGHT_APPLICATION_DELETE",
"RIGHT_APPLICATION_DEVICES_READ",
"RIGHT_APPLICATION_LINK",
"RIGHT_APPLICATION_TRAFFIC_READ",
"RIGHT_APPLICATION_DEVICES_READ_KEYS",
"RIGHT_APPLICATION_DEVICES_WRITE_KEYS",
"RIGHT_APPLICATION_ALL",
"RIGHT_APPLICATION_INFO",
"RIGHT_APPLICATION_SETTINGS_API_KEYS",
**"RIGHT_APPLICATION_TRAFFIC_DOWN_WRITE"**
]
}

root@propc:/home/robert# **ttn-lw-cli applications link set app3 --api-key** NNSXS.GIPVAVIAO77RFBOWTDO43XIZ5SDTRPDX7QKOGCQ.EJJRVXWEX7LIFSPYMT4WBPPSCIMIJEXTRMFOSRNT47RPF6KTWCJQ

{
"api_key": "NNSXS.GIPVAVIAO77RFBOWTDO43XIZ5SDTRPDX7QKOGCQ.EJJRVXWEX7LIFSPYMT4WBPPSCIMIJEXTRMFOSRNT47RPF6KTWCJQ"
}

All commands i performed are above (from gettingstarted manual).

For **TTN service cloud** we tested & developed "HTTP Integration" and works OK for common work uplink/downlink.

**I see some other issues LW-stack:**
- **TTN service cloud** seams to work very stabile almost any data passing trough but lorawan-stack missing a lot of data (device not found messages) after 2-5 data uplink (no re-join is possible - sometimes reset to factory defaults of modem and repeat all initializations steps from the begining helps).

What we are testing now is firmware update via LoraWAN so there are big data thrughput not conforming "Fair data transfers" for TTN service cloud.

adriansmares, 2019-06-25:
I've been unable to reproduce this on the latest release (which I suppose you are using since you've mentioned that you are following the getting started guide). My logs are the [here](https://pastebin.com/JPa2Cd6A).

Your answer regarding `NNSXS.GIPVAVIAO77RFBOWTDO43XIZ5SDTRPDX7QKOGCQ.EJJRVXWEX7LIFSPYMT4WBPPSCIMIJEXTRMFOSRNT47RPF6KTWCJQ` still does not answer what rights does this key have. You need two API keys - one for linking, which should have only `--right-application-link` and one for Webhooks, which should have `--right-application-traffic-read` and `--right-application-traffic-down-write`.

johanstokking, 2019-06-25:
@ecities I still see two different API keys. If you are using the one you mention in your last post for linking _and_ for sending downlink, it probably doesn't have the right rights. The one you created will have the right rights. Apart from that, there may be a formatting issue as @adriansmares points out above.

> * **TTN service cloud** seams to work very stabile almost any data passing trough but lorawan-stack missing a lot of data (device not found messages) after 2-5 data uplink (no re-join is possible - [...]

We need a bit more context for this. Best is to open a new issue. Mention MAC version, OTAA vs ABP, whether you reset nonces because (OTAA) or reset frame counters (ABP) on factory reset, etc.

ecities, 2019-06-25:
Thanks a lot 
 it looks that complete clearing all apps, gws, api-keys, devices and 
following Adrian's instructions in log file from the beginning
did the trick and **downlink working OK**

**I think adding this to getting started is crucial - it's not there :)**
`ttn-lw-cli applications api-keys create \
  --name wh-client \
  --application-id app3 \
  --right-application-traffic-read \
  --right-application-traffic-down-write`

johanstokking, 2019-06-25:
Glad to hear. And thanks for the suggestion, that's a fair point. We'll pick that up.

ecities, 2019-06-26:
I created second app,api-keys,webhook, etc for the same device (dev1) (for Firmware UPGRADE) and it have the same error (403) on server console - and access rights problems. 
**But when I use docker-compose down & up it starts working OK.
Does  webhooks operations requires server down/up or some time delay?**

johanstokking, 2019-06-26:
No, those changes are effective immediately.

ecities, 2019-06-26:
ok, thanks

ecities, 2019-06-26:
Hi,
I'm testing webhooks for LoRaWAN firmware update and it seems downlink have very small size limit
for 26B downlink data is working
for 57B  downlink is too big - got message on server console:
**WARN: Application downlink present, but the payload is too long.**

**What is the limit and? Can it be adjusted to send as big packet as possible?**
We must update at least 64kB so packet size really matters.

johanstokking, 2019-06-26:
The maximum payload size depends on the PHY (regional parameters) and data rate. Please see the LoRaWAN Regional Parameters for your version, for example [1.0.3 rev A](https://lora-alliance.org/resource-hub/lorawan-regional-parameters-v103reva).

ecities, 2019-06-27:
Johan thanks for replay and pointing out data rate but default settings of modem looks OK

I'm on Telit re866 module which is 1.0.2 ClassA/C certified & check LoRaWAN 1.0.2 for regional parameters.
I'm using Class A (unable to switch to class C)

Default settings datarate (sf12/125kHz) - rx2datarate (default as for lora 1.0.2 spec)
Testing conditions: -90dBM
I also set other datarate (sf7/125) (adoptive dr=on) kerlink femtocell shows in logs (sf7bw125) - according to lora specs max. payload size is 230/222. I tested other values with the same payload size and sf12 (not defined).

The biggest downlink size transfer I manage to send is 54B (binary) - base 64 coded (72)

johanstokking, 2019-06-27:
@ecities can you file a separate issue for this and fill out the issue template?

Let's keep this issue specific to documenting creating an API key for working with webhooks. That's now burried in the MQTT section.

ecities, 2019-06-27:
OK thanks,

johanstokking, 2019-06-27:
We did not finish documenting creating an API key for working with webhooks.

ecities, 2019-06-27:
OK, sorry wrong button  

From: Johan Stokking 
Sent: Thursday, June 27, 2019 12:54 PM
To: TheThingsNetwork/lorawan-stack 
Cc: ecities ; State change 
Subject: Re: [TheThingsNetwork/lorawan-stack] Document creating API key for webhooks (#845)

We did not finish documenting creating an API key for working with webhooks.

PR #886 (2019-07-01): Getting Started improvements

#### Summary

Closes #845 
Closes #850 

#### Changes

- Add a note regarding the API key required for Webhooks downlink queue operations.
- Add multicast device creation instructions and point to them for class C multicast.

#### Notes for Reviewers

None.
closed 2019-07-01 by adriansmares