TheThingsNetwork/lorawan-stack #849: Multicast device support in console

rvolosatovs, 2019-06-25:
#### Summary

[image]
Multicast support to Network Server was added in https://github.com/TheThingsNetwork/lorawan-stack/pull/792
Console should allow creation of multicast devices.
From the perspective of console a multicast device can be seen as just an ABP device, which does not send uplinks.
The `multicast` field on `EndDevice` proto message https://github.com/TheThingsNetwork/lorawan-stack/blob/576cac2962c6f67d2fb9de80afa8a7f7b10deb8f/api/end_device.proto#L473 has to be set to `true` on creation of the device in order for the it to correctly operate as a part of multicast group.
The value cannot be changed after the device creation.

#### Why do we need this?

To provide a nice UI for our users, who want to create multicast devices

#### What is already there? What do you see now?

No UI described in this issue

#### What is missing? What do you want to see?

UI described in this issue

#### Environment

Any

#### How do you propose to implement this?

Maybe just have a *multicast* checkbox?
Note, that OTAA devices cannot be multicast by definition.

#### Can you do this yourself and submit a Pull Request?

I'm sure @bafonins or @kschiffer will be better suited for the job.

johanstokking, 2019-06-25:
I think UX wise multicast is like ABP; it's a fixed session.

We could have three modes to create a device:
- OTAA
- ABP
- Multicast (which is like ABP but sets `multicast`)

bafonins, 2019-07-01:
Something like this?

[image]

rvolosatovs, 2019-07-01:
Yes!
Please remove/disable the *Resets Frame Counters*, since no uplinks can be sent

closed 2020-06-23 by bafonins