TheThingsNetwork/lorawan-stack #857: Collaborators can revoke all rights from other collaborators, regardless of their own rights

kschiffer, 2019-06-26:
#### Summary
A collaborator that has only the `RIGHT_{entity}_SETTINGS_COLLABORATORS` right can revoke all rights from any other collaborator of the same entity. This should, to my understanding, be limited to only the rights that the collaborator has himself. 
It is already not possible to enable a right for another collaborator that is not granted to oneself. However, revoking such rights is still possible.

#### Steps to Reproduce
1. Create a user with just the `RIGHT_{entity}_SETTINGS_COLLABORATORS` right
2. Create another user with `RIGHT_{entity}_ALL` within the same entity
3. As the first user, call the `SetCollaborator` rpc for the second user with an empty array as set of rights
4. The collaborator which had all rights is removed as a collaborator (by a user with fewer rights)

#### What do you see now?
It might be that it is indeed intended for a user with `RIGHT_{entity}_SETTINGS_COLLABORATORS` to be able to modify any right of another collaborator. However, at the moment out-of-own-scope rights can only be removed and not set, which is a weird inconsistency.

#### What do you want to see instead?
Either: restriction to modify only rights that the collaborator has himself
Or: possibility to fully modify all rights, meaning enabling and disabling them

#### How do you propose to implement this?
I think I need some input as to what the actual strategy is for these cases.

#### Can you do this yourself and submit a Pull Request?
Needs clarification.

htdvisser, 2019-06-27:
This is how we designed it. The `RIGHT_{entity}_SETTINGS_COLLABORATORS` gives someone the right to view and edit collaborators of an entity, with one constraint: it is not possible to assign someone else more rights than you have yourself.

I can't think of any realistic scenario where a collaborator would have the right to manage collaborators, but no other rights, so I don't see this edge case being a problem.

I also think it's too late to overhaul the rights even if we wanted to.

johanstokking, 2019-06-27:
>This is how we designed it.

Indeed.

I don't see this as an issue.

closed 2019-06-27 by johanstokking