TheThingsNetwork/lorawan-stack #864: Support TLS when connecting to a redis server

dblane-digicatapult, 2019-06-27:
#### Summary

Add support for using tls when connecting to redis server
...

#### Why do we need this?

Currently we can only connect to non-tls redis servers.
...

#### What is already there? What do you see now?

Currently the network server times out when trying to connect to a TLS redis server.
It appears the go library you are using for redis supports TLS.

If you try and use the standard redis connection notation e.g ```rediss://server:host``` then you get the error ```--- dial tcp: address rediss://redis:6379: too many colons in address```

...

#### What is missing? What do you want to see?

TLS support for redis
...

#### Can you do this yourself and submit a Pull Request?

Unfortunately not, I have insufficient experience with go.
...

johanstokking, 2019-06-27:
Thanks, this is a good one for the community to pick up!

kartben, 2020-11-09:
FWIW I just learned this is supported in v3 (and I just tested it successfuly) thanks to the `redis.tls.require` configuration flag.
See https://thethingsstack.io/reference/configuration/the-things-stack/#redis-options for more Redis options.

johanstokking, 2020-11-09:
Awesome! So we can close this?

closed 2020-11-09 by johanstokking