TheThingsNetwork/lorawan-stack #866: Webhook downlink limit up to 54Bytes (binary) - payload is to long and ignored by server

ecities, 2019-06-27:
#### Summary

**I can't sent downlink with data payload size larger than 54B (binary) / base 64 coded (72B)
With higher size server return message:
WARN Application downlink present, but the payload is too long, Inform Application Server ack=true........**

I'm on Telit re866 module which is 1.0.2 ClassA/C certified & check LoRaWAN 1.0.2 for regional parameters.
I'm using Class A 

Default settings datarate (sf12/125kHz) - rx2datarate (default as for lora 1.0.2 spec)
Testing conditions: -90dBm

I also set other Datarates eg. (sf7/125) => Kerlink femtocell GW shows in logs (sf7bw125) 
As I understood according to lora specs max. payload size is 230/222. I tested other values with the same payload size and sf12 the same results.

...

#### Steps to Reproduce

1. ...   curl http://localhost:1885/api/v3/as/applications/ap2/webhooks/fwup/devices/dv1/down/push -X POST   -H 'Authorization: Bearer NNSXS.CLCIYOYY********'  --data '{"downlinks":[{"frm_payload":"AAECAwQFBgcICQoLDA0ODxAREhMUFRYXGBkaGxwdHh8gISIjJCUmJygpKissLS4vMDEyMzQ1Njc4OTo7PD0+P0BBQkNERUU=","f_port":15,"priority":"HIGHEST"}]}'

2. Data 71B (binary) ... Webhook  Status (200) - OK
3. ... Server console message after a while: **WARN Application downlink present, but the payload is too long, Inform Application Server ack=true........**
4. Data is not received by Node 

#### What do you see now?

 **WARN Application downlink present, but the payload is too long, Inform Application Server ack=true........**
...

#### What do you want to see instead?

...

#### Environment

Ubuntu 16.4/FF/Kerlink Femtocell/Telit RE866
...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2019-06-27:
- Which PHY are you in?
- Which data rate are you using for uplink when you see that error?

ecities, 2019-06-27:
Node Telit Re866+uC (EU863-870 - lorawan 1.0.2 Class A)
pkt info from femtocell kerlink + spf (tx): for data limited to 52B (binary)
Local network ( Kerlink Ethernet<=>PC Etnernet )
Modulation: LORA
codr: 4/5
imme: false
rfch: 0
powe: 14
ipol: true
ncrc: true
size: 65
datr: sf7bw125

RSSI (rx): -80dBm

**The same settings for Payload length: 62B (binary)**
I scan all posibilities for DataRate (all fails).
Uplink datarates tested (data not received on the node + WARN "payload too long" on server):
SF7bw125 ---
SF12bw125 ---
SF11bw125 --
SF10bw125 --
SF9bw125 --
SF8bw125 --
SF7bw250 --- error join
FSK -- error join

rvolosatovs, 2019-06-27:
This happens because the configured Rx2 data rate is SF12BW125(DR0). In class A NS currently schedules downlink such, that it complies with both Rx1 and Rx2 parameters. According to the table from the regional parameters spec, maximum size of FRMPayload in DR0 that can be used for Rx2(in case FOpts are absent) is 51, which is less than what you're trying to do.
[image]

@johanstokking I think we should change this behavior and NS should in fact separate Rx1 and Rx2 scheduling requests to GS if the payload does not fit in either.

johanstokking, 2019-06-27:
@rvolosatovs yes, it should definitely do that.

ecities, 2019-06-28:
Is there some fast workaround? Can we enforce data rate in **downlink json configuration?**

We need to send hundreds downlink packets for FW upgrade:
-  about 200B each (best case)
- about 70B each   (worst case)

We consider to add additional LoRaWAN gateway just for firmware upgrade to establish "good range" and different Network/Application Servers (for production environment).

johanstokking, 2019-06-28:
> Is there some fast workaround? Can we enforce data rate in **downlink json configuration?**

You can change the Rx2 parameters for the end device, see `ttn-lw-cli dev set --help`, with which you can set the Rx2 data rate to a high value, i.e. 5 (SF7BW125 in EU868).

ecities, 2019-06-28:
Thanks @johanstokking this solution fix the issue.

Currently I upgrade firmware for LoRaWAN module and have Class C available.

closed 2019-07-04 by rvolosatovs

johanstokking, 2019-07-05:
@ecities this lands in the next release, please star this repository or subscribe to release notifications.

viethoa14, 2021-01-19:
hi 
now we have problem following:
"ERROR Generated downlink payload size does not fit neither RX1, nor RX2, skip class A downlink slot band_id=AS_923 dev_addr=FC005138 device_class=CLASS_A..."
we use sf10 for defaul, i dont't understand why

rvolosatovs, 2021-01-19:
Hi @viethoa14 
The information you've provided is not sufficient to debug this issue. If you are TTI customer, please reach out to our support team via the ticketing system.
If you're an open source user, please provide the following clearly reproducing the issue:
- Stack log
- Device event log