TheThingsNetwork/lorawan-stack #869: Fix collaborator rights check for data traffic

bafonins, 2019-06-27:
#### Summary

Application/Gateway collaborator cannot observe events when having sufficient rights

#### Steps to Reproduce

1. Create a new application
2. Add a collaborator and assign all possible application rights:
```
"RIGHT_APPLICATION_DELETE"
"RIGHT_APPLICATION_DEVICES_READ"
"RIGHT_APPLICATION_DEVICES_READ_KEYS"
"RIGHT_APPLICATION_DEVICES_WRITE"
"RIGHT_APPLICATION_DEVICES_WRITE_KEYS"
"RIGHT_APPLICATION_INFO"
"RIGHT_APPLICATION_LINK"
"RIGHT_APPLICATION_SETTINGS_API_KEYS"
"RIGHT_APPLICATION_SETTINGS_BASIC"
"RIGHT_APPLICATION_SETTINGS_COLLABORATORS"
"RIGHT_APPLICATION_TRAFFIC_DOWN_WRITE"
"RIGHT_APPLICATION_TRAFFIC_READ"
"RIGHT_APPLICATION_TRAFFIC_UP_WRITE"
```
3. Login as the collaborator
4. Observe no events being shown for the application

**However, this works if the collaborator gets the `RIGHT_APPLICATION_ALL` right.**

#### What do you see now?

No events being shown for the collaborator with all necessary rights
```
{
   "error":{
      "grpc_code":7,
      "http_code":403,
      "message":"error:pkg/auth/rights:insufficient_application_rights (insufficient rights for application `new-app`)",
      "http_status":"Forbidden",
      "details":[
         {
            "@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace":"pkg/auth/rights",
            "name":"insufficient_application_rights",
            "message_format":"insufficient rights for application `{uid}`",
            "attributes":{
               "uid":"new-app"
            },
            "correlation_id":"37879009e0164eea8e30fed8af74197c"
         }
      ]
   }
}
```

#### What do you want to see instead?

No error and successfully opened events stream for the collaborator with the `RIGHT_{APPLICATION|GATEWAY}_TRAFFIC_READ` right

#### Environment

AS

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

no

johanstokking, 2019-06-27:
This is caused by this;

https://github.com/TheThingsNetwork/lorawan-stack/blob/68708557cb13a3cdfa8556c8c1c814074bcff06a/pkg/events/grpc/server.go#L85

I think we have to filter the event stream by checking item-by-item whether there are sufficient rights for the event type.

I assume that @htdvisser already has some ideas for this.

johanstokking, 2019-07-01:
I can imagine we'll get some filter that switches on the event type and potentially subfield types. For example, if the event payload is of type `*ttnpb.ApplicationUp`, require `RIGHT_APPLICATION_TRAFFIC_READ`, with fallback to the entity type's `ALL` right.

htdvisser, 2019-07-01:
As @johanstokking wrote last week, the rights checks in this RPC indeed have to change a bit. The check will have to be split into two parts: first, we need to check that the caller has _any_ right on the requested entity, and second, we need to check which rights are required for each event.

I don't think it's a good idea to switch event (payload) visibility based on payload type. We shouldn't have that kind of logic in `pkg/events`, because then we'll have to edit `pkg/events` whenever we add a new event type or change the payload of an event.

I think it's better to make the required rights part of the event definition, perhaps even the event itself, so that the rights are carried over gRPC or pubsub. If we only make it part of the definition, the event server binary (currently ttn-lw-stack when it's running the console) will have to import all packages that define events. If we make it part of the events themselves, we don't need to do that, which gives us more flexibility if we want to switch out the events server with a separate microservice.

Then we have the question of how required rights are determined for events. I think it would be nice to have the possibility of separating the right to see that an event took place and the right to see the actual event payload.

Another question is how rights should be checked. Is is a "caller has any of the required rights" or "caller has all of the required rights". If we're going to add a list of required rights to each event, this choice needs to be made, as there won't be a possibility of adding custom logic for this.

Finally, a complicating factor is that currently events can be linked to multiple identifiers (for example, the `oauth.authorize` event is linked to a User and a Client). Since this isn't very common, now may be a good time to decide that we no longer want to support this. That would make the implementation of rights checks a lot simpler.

johanstokking, 2019-07-01:
> I think it's better to make the required rights part of the event definition, perhaps even the event itself, so that the rights are carried over gRPC or pubsub. If we only make it part of the [...]

Good idea.

> Then we have the question of how required rights are determined for events. I think it would be nice to have the possibility of separating the right to see that an event took place and the right to see the actual event payload.

Can you give an example use case? Like, "the gateway is receiving messages, but you need rights to see them"? Not sure if we need this (already).

> Another question is how rights should be checked. Is is a "caller has any of the required rights" or "caller has all of the required rights". If we're going to add a list of required rights to each event, this choice needs to be made, as there won't be a possibility of adding custom logic for this.

I think it would be any. If we need an intersection, we may need to consider introducing a new right. At this point, I can't think of a use case.

> Finally, a complicating factor is that currently events can be linked to multiple identifiers (for example, the `oauth.authorize` event is linked to a User and a Client). Since this isn't very [...]

Didn't we decide to stop supporting this before and/or related to https://github.com/TheThingsNetwork/lorawan-stack/pull/685 ? In this particular case, can't we fire two events?

htdvisser, 2019-07-01:
An example use case could indeed be some kind of per-message billing integration that hooks into application data events. For billing it only needs to know *that* there was an uplink/downlink, but it doesn't need to see the contents.

#685 indeed removed the "plural" from `ttnpb.Identifiers`, but multiple identifiers are still supported where it made sense. The rights hook and events package are places where it is likely that a single request/event is related to multiple entities. For the rights hook this is quite essential, because (in the future) we need to check that for example a caller of a SetApplicationCollaborators RPC has the right to edit collaborators of the application, but also the right to add(invite) the collaborating organization/user to an entity. For the OAuth Authorize we can indeed fire multiple events, but then they will have different timestamps and IDs while they're still the same event. If we can live with that, we can do it.

PR #1038 (2019-07-24): Add Rights to Events

#### Summary

This Pull Request adds a `visibility` field to our events, which will carry the rights that are required to see each event. 

This Pull Request closes #869 and closes #904.

#### Changes

- Added `visibility` to Event proto
- Use visibility in pkg/events
- Refactored rights checking in pkg/events/grpc
    - We now check that the caller has at least one right on the entities when the call starts
	- for this we're using the now exported errors in pkg/auth/rights
- Set visibility on existing events
- Fixed missing event in IS
- Fixed missing collaborator identi [... cut]
closed 2019-07-24 by htdvisser