TheThingsNetwork/lorawan-stack #873: Send Time Synchronization Data from Network Server to nodes

ecities, 2019-06-28:
#### Summary

Is there a way of time synchronization of nodes from the Network Server?

We have tested downlink "set time info packet" (html integration and webhook) but it always arrives next upload time (even in local network no matter downlink priority is set) - so is very inacurate.
...

#### Why do we need this?

To synchronize clock after device reboots?
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

Send current time of Network Server upon query from node ...

#### Can you do this yourself and submit a Pull Request?

...

KrishnaIyer, 2019-06-28:
@ecities: What level of precision are you aiming for? Device level time sync is difficult to achieve without Class B gateway beacons.

johanstokking, 2019-06-28:
@ecities please see https://github.com/TheThingsNetwork/lorawan-stack-example-clock for an example.

When using class A, by nature, the downlink slot depends on the uplink. When using class B or C, that is not the case. TTN V3 supports absolute time scheduling on class C, either using GPS synced gateways or server time using gateway RTT.

johanstokking, 2019-06-28:
@ecities you may also be interested in an application layer protocol for this, which is device initiated, see https://lora-alliance.org/resource-hub/lorawan-application-layer-clock-synchronization-specification-v100

ecities, 2019-06-28:
@KrishnaIyer We need just set internal clock. Accuracy is not too critical ~1..5 sec and it seems to be better than sending packets with time from external (integrated) server.

Our controllers supports calendar-scheduler. Microcontoller doesn't have RTC and we want to go into deep sleep for some variants of software.
Our LORAWAN modem is Class A + C, but I see it can't be switch to class C in our firmware rev. (we are on hot line with manufacture for firmware upgrade). On the other hands The LORAWAN modem do not have "Time commands" as usually GSM modems do.
@johanstokking thanks for the link, I will read it asap.

johanstokking, 2020-02-07:
We'll be adding support for ALCS and class B.

closed 2020-02-07 by johanstokking