TheThingsNetwork/lorawan-stack #876: Insufficient rights as admin to manage collaborators

johanstokking, 2019-06-28:
#### Summary

Insufficient rights as admin to manage collaborators

#### Steps to Reproduce

1. Login as admin
2. Create a new organization that is owned by user A
3. Add collaborator user B to this new organization

#### What do you see now?

```
$ ttn-lw-cli organizations members set --organization-id test-org --user-id user-b --right-application-all --right-gateway-all
  WARN Finished unary call                      duration=46.145608ms error=rpc error: code = PermissionDenied desc = error:pkg/auth/rights:insufficient_organization_rights (insufficient rights for organization `test-org`) error_correlation_id=4c998dbee1aa4d4ab45b184b7e12cdd6 error_name=insufficient_organization_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=grpc
error:pkg/auth/rights:insufficient_organization_rights (insufficient rights for organization `test-org`)
    uid=test-org
    correlation_id=4c998dbee1aa4d4ab45b184b7e12cdd6
```

#### What do you want to see instead?

Admins should be able to do this stuff

#### How do you propose to implement this?

Admins should have implied right `RIGHT_ORGANIZATION_SETTINGS_MEMBERS`

#### Can you do this yourself and submit a Pull Request?

No

johanstokking, 2019-07-01:
Server logs:

```
stack_1      |  DEBUG Run database query                       duration=1ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "access_tokens"  WHERE ("access_tokens"."token_id" = $1) ORDER BY "access_tokens"."id" ASC LIMIT 1 request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=oauth_store.go:253 values=[LHUDIKBIOBGWNOC4EF5FFLVW6YPSUPCND6GEFPI]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND (("id" IN ($1))) ORDER BY "clients"."id" ASC request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=oauth_store.go:253 values=[4ca69399-8c66-475c-9db5-400449aa686a]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "users"  WHERE "users"."deleted_at" IS NULL AND (("id" IN ($1))) ORDER BY "users"."id" ASC request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=oauth_store.go:253 values=[cdb9f289-5329-44b1-8182-dea91448a267]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "accounts"  WHERE "accounts"."deleted_at" IS NULL AND (("account_id" IN ($1) AND "account_type" = $2)) ORDER BY "accounts"."id" ASC request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=oauth_store.go:253 values=[cdb9f289-5329-44b1-8182-dea91448a267 user]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT users.id, users.created_at, users.updated_at, admin, state, primary_email_address_validated_at FROM "users" LEFT JOIN accounts ON accounts.account_type = $1 AND accounts.account_id = users.id WHERE "users"."deleted_at" IS NULL AND ((accounts.uid = $2)) ORDER BY "users"."id" ASC LIMIT 1 request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=user_store.go:120 values=[user $USERNAME]
stack_1      |  DEBUG Run database query                       duration=1ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "accounts"  WHERE "accounts"."deleted_at" IS NULL AND (("account_id" IN ($1) AND "account_type" = $2)) ORDER BY "accounts"."id" ASC request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=user_store.go:120 values=[cdb9f289-5329-44b1-8182-dea91448a267 user]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT id, created_at, updated_at, client_id, state FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND ((client_id = $1)) ORDER BY "clients"."id" ASC LIMIT 1 request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=client_store.go:112 values=[cli]
stack_1      |  DEBUG Run database query                       duration=2ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT organizations.id, organizations.created_at, organizations.updated_at FROM "organizations" LEFT JOIN accounts ON accounts.account_type = $1 AND accounts.account_id = organizations.id WHERE "organizations"."deleted_at" IS NULL AND ((accounts.uid = $2)) ORDER BY "organizations"."id" ASC LIMIT 1 request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=organization_store.go:116 values=[organization $ORGNAME]
stack_1      |  DEBUG Run database query                       duration=1ms grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=db query=SELECT * FROM "accounts"  WHERE "accounts"."deleted_at" IS NULL AND (("account_id" IN ($1) AND "account_type" = $2)) ORDER BY "accounts"."id" ASC request_id=01DEPCFBWQHBT2RJV4DBBC9RJE rows=1 source=organization_store.go:116 values=[46eaa563-dfbc-476b-b2d2-d705b72efa57 organization]
stack_1      |   WARN Finished unary call                      duration=29.948636ms error=error:pkg/auth/rights:insufficient_organization_rights (insufficient rights for organization `$ORGNAME`) error_correlation_id=076d9a3e7308447ea56cc7674b34c91b error_name=insufficient_organization_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=SetCollaborator grpc_service=ttn.lorawan.v3.OrganizationAccess namespace=grpc request_id=01DEPCFBWQHBT2RJV4DBBC9RJE
stack_1      |  DEBUG transport: loopyWriter.run returning. connection error: desc = "transport is closing" namespace=grpc
```

htdvisser, 2019-07-09:
This actually works as intended. 

We made the decision that admins don't have `_KEYS` or `_ALL` rights on entities that they don't actually have rights to. Since users can never assign rights to collaborators that they don't already have themselves, admins can't assign these rights to collaborators unless they actually have those rights.

closed 2019-07-10 by htdvisser