TheThingsNetwork/lorawan-stack #881: Console doesn't respect disabled components

johanstokking, 2019-06-28:
#### Summary

Console doesn't respect disabled components when working with devices.

#### Steps to Reproduce

1. Disable some components, i.e. NS and AS, but keep JS enabled, see example config below
2. Create a device through CLI
3. Try to get the device in the Console

Example config:

```
TTN_LW_CONSOLE_UI_AS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_AS_ENABLED="false"
TTN_LW_CONSOLE_UI_GS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_GS_ENABLED="false"
TTN_LW_CONSOLE_UI_IS_BASE_URL="https://<snip>/api/v3"
TTN_LW_CONSOLE_UI_IS_ENABLED="true"
TTN_LW_CONSOLE_UI_JS_BASE_URL="https://<snip>/api/v3"
TTN_LW_CONSOLE_UI_JS_ENABLED="true"
TTN_LW_CONSOLE_UI_NS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_NS_ENABLED="false"
```

Browser receives this correctly:

```js
<script>window.APP_ROOT="/console";window.ASSETS_ROOT="/assets";window.APP_CONFIG={"language":"en","is":{"enabled":true,"base_url":"https://<snip>/api/v3"},"gs":{"enabled":false,"base_url":"http://localhost:1885/api/v3"},"ns":{"enabled":false,"base_url":"http://localhost:1885/api/v3"},"as":{"enabled":false,"base_url":"http://localhost:1885/api/v3"},"js":{"enabled":true,"base_url":"https://<snip>/api/v3"}};window.SITE_NAME="The Things Network Stack for LoRaWAN";window.SITE_TITLE="Console";window.SITE_SUB_TITLE="The official configuration platform for The Things Network";</script>
```

#### What do you see now?

The browser shows two 404s to:

- https://<snip>/ns/applications/test-app/devices/test-dev?field_mask=session,version_ids,frequency_plan_id,mac_settings.resets_f_cnt,supports_class_c,supports_join,lorawan_version,lorawan_phy_version
- https://<snip>/as/applications/test-app/devices/test-dev?...

Note the missing `/api/v3` here

#### What do you want to see instead?

No tries to AS and NS endpoints with this config

#### How do you propose to implement this?

No idea

#### Can you do this yourself and submit a Pull Request?

No but can test

johanstokking, 2019-08-08:
This is not entirely addressed by #994;

Link, Payload Formatters (app and device level) and Integrations are AS-specific.

kschiffer, 2019-09-24:
Closing this for #1200, please refer to this issue.

PR #994 (2019-07-15): Add proper handling of disabled components to JS SDK
#### Summary
Closes #881.
This PR is the first step towards proper support of different stack component setups. It introduces a couple of fixes and enhancements to ensure the SDK can handle arbitrary setups in terms of what components (AS, IS, JS, NS, etc) are enabled or disabled. As a result, the console will be able to receive proper errors for queries to disabled components.

#### Changes
- Fix an issue where passed `stackConfig` keys set as `undefined` would still be mapped to axios instances without a `baseUrl` (leading to requests with [... cut]