TheThingsNetwork/lorawan-stack #887: CLI panic on listing organizations without passing user ID

johanstokking, 2019-07-01:
#### Summary

CLI panics when listing organizations without specifying a user ID.

#### Steps to Reproduce

1. Login
2. List organizations with `$ ttn-lw-cli organizations ls`

#### What do you see now?

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x1c6f159]

goroutine 1 [running]:
go.thethings.network/lorawan-stack/cmd/ttn-lw-cli/commands.glob..func71(0x2b6a040, 0x2bbef90, 0x0, 0x0, 0x0, 0x0)
	/Users/johan/dev/src/lorawan-stack/cmd/ttn-lw-cli/commands/organizations.go:78 +0x1d9
github.com/spf13/cobra.(*Command).execute(0x2b6a040, 0x2bbef90, 0x0, 0x0, 0x2b6a040, 0x2bbef90)
	/Users/johan/go/pkg/mod/github.com/spf13/cobra@v0.0.3/command.go:762 +0x465
github.com/spf13/cobra.(*Command).ExecuteC(0x2b6c640, 0x0, 0x0, 0x1c814cc)
	/Users/johan/go/pkg/mod/github.com/spf13/cobra@v0.0.3/command.go:852 +0x2ec
github.com/spf13/cobra.(*Command).Execute(...)
	/Users/johan/go/pkg/mod/github.com/spf13/cobra@v0.0.3/command.go:800
main.main()
	/Users/johan/dev/src/lorawan-stack/cmd/ttn-lw-cli/main.go:27 +0x32
```

#### What do you want to see instead?

No panic, but maybe a nice error that there's no user ID passed (see below)

#### Environment

Current `master`

#### How do you propose to implement this?

`getUserID()` may return `nil` when there's no `user-id` flag set or if there's no single argument.

We either need to fallback to the "current user ID" if nothing is specified, or requiring the user to specify their own user ID on these calls.

If you're in doubt, ask @htdvisser as cli code owner

#### Can you do this yourself and submit a Pull Request?

Will review

furtiman, 2019-07-17:
@htdvisser can you give a hint on how can I fetch the current user ID? When I log in and it is my session, where is it stored and how to get it?

I assume it is somewhere in the `auth cache/cookie`, but I am having issues with how to fetch it in the command runtime.

I want to use that to send a `List` request if no user Id is specified, as you mentioned [here](https://github.com/TheThingsNetwork/lorawan-stack/pull/993#issuecomment-511771675)

htdvisser, 2019-07-17:
The CLI doesn't have the current user ID, and it may not even be logged in as a user at all (it can also be logged with any API key).

I proposed an alternative fix in #1010

PR #1010 (2019-07-18): Give OrganizationOrUserIdentifiers() pointer receiver

#### Summary

This Pull Request closes #887 with the changes I proposed in https://github.com/TheThingsNetwork/lorawan-stack/pull/993#pullrequestreview-261725926.

#### Changes

- Gave `OrganizationOrUserIdentifiers()` a pointer receiver
- Made them return `nil` if receiver `nil`
- Updated usage

#### Release Notes

- Fixed crash when running some `ttn-lw-cli organizations` commands without `--user-id` flag.
closed 2019-07-18 by htdvisser