TheThingsNetwork/lorawan-stack #889: Better CLI multicast support.

adriansmares, 2019-07-01:
#### Summary

#792 added support for multicast devices in the Network Server by adding the `Multicast` flag. This flag is supported currently in the CLI, under the form of `--multicast`, but requires the `--abp` flag to be set as well, in order to set `SupportsJoin` to false. It is however rather unintuitive that the device is both ABP and Multicast at the same time.

#### Why do we need this?

In order to avoid any confusion regarding the device types in the CLI.

#### What is already there? What do you see now?

`--multicast` must be used with `--abp`, otherwise the creation will fail, as expected.

#### What is missing? What do you want to see?

`SupportsJoin` should be false if the device is either `--abp` or `--multicast`, in order to drop `--abp` from the device creation command.

#### Environment

Not really important, but 861fe5eaec412b8da0ff47cdb431b41e49a30b50.

#### How do you propose to implement this?

[This](https://github.com/TheThingsNetwork/lorawan-stack/blob/861fe5eaec412b8da0ff47cdb431b41e49a30b50/cmd/ttn-lw-cli/commands/end_devices.go#L328-L367) needs to be changed to include `--multicast` as well.

#### Can you do this yourself and submit a Pull Request?

Yes, but if anyone else wants to pick it up feel free to do so.

htdvisser, 2021-03-26:
Is this still relevant?

adriansmares, 2021-03-26:
No
https://github.com/TheThingsNetwork/lorawan-stack/blob/8e82540e590f3b8ad7bdfa5045c4abeb095ea112/cmd/ttn-lw-cli/commands/end_devices.go#L376-L377

closed 2021-03-26 by adriansmares