TheThingsNetwork/lorawan-stack #890: Issue template for questions

htdvisser, 2019-07-01:
#### Summary

I propose to add an issue template for questions.

#### Why do we need this?

In #871 and #873 we see that the current issue templates are not applicable to questions.

#### What is already there? What do you see now?

We have issue templates for bugs and features. We don't have an issue template for questions, but we also don't accept issues that don't follow an issue template.

#### What is missing? What do you want to see?

I think it would be good to create an issue template for questions. This template can contain a checklist of other places to look first

#### How do you propose to implement this?

- Summary
- Why are you asking this question?
- Steps to reproduce / Steps you took
  - [ ] Did you search the documentation?
  - [ ] Did you search the forum?
  - [ ] Did you search existing issues?
- What is already there? What did the documentation/forum/issue say?
- What is missing to answer your question?
- ...

#### Can you do this yourself and submit a Pull Request?

Let's discuss first

KrishnaIyer, 2019-07-01:
so the "action" would be that we update documentation directly (if that's the only change) referencing this issue and open another one if this results in a bug fix/feature request?

htdvisser, 2019-07-01:
Exactly. The result of a "question" issue should usually be a change in our documentation, or an addition to our FAQ. 

I don't know if it makes sense to copy-paste a "question" issue into a "bug_report" issue. Maybe we should just add the relevant sections in comments (or edit the issue) if we decide to turn a question into a bug report or feature.

Sypheos, 2019-07-01:
Agree but that should also apply to the forum and slack.

In essence all information should be in  the repo documentation, if some information is reference on the forum, issues or slack then we failed to make a documentation. 
Question or issue raised on slack, forum should make up for change in doc unless no effort were taken to look for the information.

Instead having of list of "did you look there", I would add a link to a query on github / doc (once search is impl) . Looking for information in github issues can be tedious and the forum shouldn't hold information/explanation the doc doesn't have. A link to an already established query (like looking for bugs) will help. Adding a know bugs section to the docs can also be envisaged.

If a question put light on a bug it is up to the dev to qualify it as a bug (and edit the issue as a bug report) or open another issue if the only part of the question involve a bug.

htdvisser, 2019-07-01:
I completely agree that we should better monitor the Forum and Slack and turn (good) questions into documentation improvements (at least for the [v3 category](https://www.thethingsnetwork.org/forum/c/network-and-routing/v3) and [lorawan-stack channel](https://thethingsnetwork.slack.com/messages/CFVF7R4AH/)). @Sypheos how would you see that working in practice?

The goal of the "have you looked here" was more of a filter for people submitting issues before doing any searching at all. The next section is more useful, there we can ask the issue submitter to link to (or quote) relevant forum posts, slack messages etc.

johanstokking, 2019-07-01:
I think this is a good idea. Let's keep the scope of this issue and issue template really a new question.

As for documentation improvements, i.e. informal answers on a forum and Slack that we want part of the documentation, we may consider an additional "documentation request" issue template, which is minimal; it basically describes the documentation gap and the link to where information is now (i.e. a link to the forum/Slack message).

Sypheos, 2019-07-01:
For the question template imo:
**What are looking for / to do ? Why ?**
**Where did you look ?**
_Add any webpage, issues, query againts github, the docs, forum are welcomed_
**Why was not satisfying ?**
_404 not found is a legit anwser_

Label: question

if there are no resources to answer the question, open a "documentation request" issue.

johanstokking, 2019-07-01:
We should really make sure that any answers end up in documentation, otherwise the closed issues here becomes the V3 knowledgebase, and we should avoid that. So maybe we should even skip the "question" template and go directly for "documentation request".

Also, when use of this repository grows, and this ends up the go-to place for asking questions and we get a hard time moderating that (i.e. pointing to docs and closing issues), we may regret that in the future. Note that bigger repositories like [VS Code](https://github.com/microsoft/vscode) communicate that you should not be asking questions quite aggressively:

[image]

[image]

In short, let's consider;
- Not allow questions at all, only "documentation request" issues. Questions are still valid, but only if it's indeed not documented otherwise it's not a valid documentation request and we close it
- Direct questions to the docs (when we have them) and forums

Sypheos, 2019-07-02:
Documentation request are in fact "Questions" since they were looking for something but couldn't find it. At the start they have a question that needs an answer. 
To go in that direction it would make more sense to simply deny any documentation request and questions via issues on github.

If someone want to ask a question they go to slack or the forum where someone for the core team or moderation can turn their question in a documentation PR / bug issue / feature request issue.

htdvisser, 2019-07-09:
Decision: We'll go with https://github.com/TheThingsNetwork/lorawan-stack/issues/890#issuecomment-507324845. Question template that directs to the forum, and a Documentation Request template

PR #1301 (2019-09-20): Add issue template for questions and doc requests
#### Summary

Close #890 by introducing two issue templates.
...

#### Changes
Added to issue template. One to raise documentation and one redirecting questions to the forum and slack.
closed 2019-09-20 by htdvisser