TheThingsNetwork/lorawan-stack #907: Increase page size of entity tables to 10

kschiffer, 2019-07-02:
#### Summary
The entity tables currently have a page size of 5, which is a bit low. 10 is a better value that will fill a usual browser window (16:9) better.

#### Why do we need this?
Improve UX. Page sizes that are too small are hard to navigate through. 10 is in my view a good measure of cognitive load per page, and ease of navigation. Whether 10 is indeed the optimal default needs some progressive insight.

#### What is already there? What do you see now?
- Functionality to set the page size in our `<Pagination />` component.
- Various default value constants for page sizes, set to 5

#### What is missing? What do you want to see?
A unified default value constant of 10 items per page for all entity tables.

#### Environment
Console.

#### How do you propose to implement this?
Trivial.

#### Can you do this yourself and submit a Pull Request?
Yes.

johanstokking, 2019-07-02:
I'm opting for 20, 25 or 30, potentially making the table a bit more condense.

kschiffer, 2019-07-03:
Ok, let's use 20 then. Anything higher makes the page already quite long. We can later provide a `Show all items` button.

PR #1020 (2019-07-22): console: Include page-size constants for tables

#### Summary

Close #907 

#### Changes

- Include constants file `pkg/webui/console/constants/page-sizes.js` containing two constants `TABLE_SIZE_SMALL` = 5 and `TABLE_SIZE_LARGE` = 20
- Include and Applied constants to tables in the following files 
  - pkg/webui/console/views/application-api-keys-list/index.js
  - pkg/webui/console/views/application-collaborators-list/index.js
  - pkg/webui/console/views/application-overview/index.js
  - pkg/webui/console/views/applications-list/index.js
  - pkg/webui/console/views/device-list/index.js
  - pk [... cut]
closed 2019-07-22 by mjamescompton