TheThingsNetwork/lorawan-stack #911: Flaky NS test

johanstokking, 2019-07-02:
#### Summary

Flaky NS test

#### Steps to Reproduce

Non-deterministic

#### What do you see now?

[flaky-ns-test.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3350742/flaky-ns-test.txt)

#### What do you want to see instead?

No errors

#### Environment

Travis CI

#### How do you propose to implement this?

No idea. Doesn't seem to be related to flow testing timing though.

#### Can you do this yourself and submit a Pull Request?

No

rvolosatovs, 2019-07-02:
@johanstokking the log you added seems to be trimmed at
```
  WARN grpc: addrConn.createTransport failed to connect to {in-process 0  <nil>}. Err :connection error: desc = "transport: Error while dialing context deadline exceeded". Reconnecting... namespace=grpc test_name=TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2
  WARN grpc: addrConn.createTransport failed to connect to {in-proces
```

Overall, it looks like the issue is again with the Travis being slow and exceeding the timeout.

johanstokking, 2019-07-03:
Yes it just ended there

johanstokking, 2019-07-03:
Seeing this more often than not on my PRs so please fix

rvolosatovs, 2019-07-03:
https://github.com/TheThingsNetwork/lorawan-stack/pull/912/commits/46537a32c6a07ef9f9734d6af811c958452c5176 should fix it once and for all, if not I'll investigate if there is something else causing this

johanstokking, 2019-08-22:
[log.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3529780/log.txt)

rvolosatovs, 2019-09-11:
Here we go again...
```
--- FAIL: TestFlow (0.00s)

    --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue (2.34s)

        --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2 (2.34s)

            networkserver_flow_test.go:153: Cluster.GetPeer called

            networkserver_flow_test.go:153: ApplicationAccess.ListRights called

            networkserver_flow_test.go:159: Cluster.GetPeer called

            networkserver_flow_test.go:159: ApplicationAccess.ListRights called

            networkserver_flow_test.go:185: Device created

            networkserver_flow_test.go:497: Device successfully joined. DevAddr: C03C6F23

            --- FAIL: TestFlow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2/uplink (1.54s)

                networkserver_flow_test.go:581: Duplicate HandleUplink returned <nil>

                networkserver_flow_test.go:573: HandleUplink returned rpc error: code = NotFound desc = error:pkg/networkserver:device_not_found (device not found)

                networkserver_flow_test.go:616: Uplink handling failed with a not-found error. The join-accept was scheduled, but the new device state most probably had not been written to the registry on time, retry.

                networkserver_flow_test.go:573: HandleUplink returned <nil>

                networkserver_flow_test.go:581: Duplicate HandleUplink returned <nil>

                doc.go:66: 

                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/networkserver_flow_test.go:653

                    Expected the container ([]string) to contain: 'GsNs-2' (but it didn't)!

                networkserver_flow_test.go:695: Cluster.GetPeer called

                networkserver_flow_test.go:705: Cluster.GetPeer called

                networkserver_util_test.go:907: Cluster.Auth called

                doc.go:66: 

                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/networkserver_flow_test.go:718

                    Expected the container ([]string) to contain: 'GsNs-2' (but it didn't)!

                doc.go:66: 

                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/networkserver_flow_test.go:715

                    Expected: true

                    Actual:   false

FAIL
```

PR #955 (2019-07-09): Fix Network Server flow test

#### Summary

Closes #911 (hopefully)

Not sure if the custom deduplication was causing the issue or not, however, if it was, then this PR will fix the issue - if it wasn't then we at least are going to see what actually is going wrong.

#### Changes

- Use default deduplication/cooldown window logic in flow tests
PR #1289 (2019-09-03): Account for the race condition in NS flow test

#### Summary

Closes #911 once and for all.

#### Changes

- Account for the possible race-condition between join state of the device being written to the registry and uplink being handled in the NS flow test without sacrificing the speed.

#### Notes for Reviewers

Whenever the race condition occurs test case now passes with a nice
```
PASS |             Flow/Redis_registry/Redis_downlink_task_queue/Class_A/OTAA/EU868/1.0.2/uplink (0.20s)
     |                 networkserver_flow_test.go:581: Duplicate HandleUplink returned <nil>
     |          [... cut]