TheThingsNetwork/lorawan-stack #918: Improve Network Server test coverage and consistency

rvolosatovs, 2019-07-02:
#### Summary

Improve Network Server test coverage and consistency

#### Why do we need this?

To ensure correctness

#### What is already there? What do you see now?

Coverage of `73.3%`

#### What is missing? What do you want to see?

Coverage of `90%` or more

#### How do you propose to implement this?

Add test cases for uncovered parts.

#### Can you do this yourself and submit a Pull Request?

Yes

nejraselimovic, 2021-06-07:
@htdvisser #960 looks the same to me, can we close this issue? If that's not the case, can you explain what's the difference here?

htdvisser, 2021-06-08:
Yeah, that's pretty much the umbrella issue for this one. I don't think this one adds a lot of value over #960, so we can close this one and track progress there.

closed 2021-06-08 by htdvisser