TheThingsNetwork/lorawan-stack #922: Use RSIG instead of RSSI/SNR when present

Sypheos, 2019-07-03:
#### Summary
Reports shows that more multi antenna gateways are starting to use the `rsig` array to send the signal metadata. We need to make sure we handle those values are present .
...

#### Why do we need this?
Default RSSI and SNR field would be empty when RSIG is present. If we don't handle RSIG the network will have uplinks with signal metadata set to 0. This will cause problem with ADR, device monitoring and location services.
...

#### What is already there? What do you see now?
https://github.com/TheThingsNetwork/lorawan-stack/blob/78ae6802c85d2f2e052567a30ffb64906d964691/pkg/ttnpb/udp/packet_data.go#L43

#### What is missing? What do you want to see?
Test or logs indicating the correct handling of uplinks packets with RSIG.

#### Environment
Gateway Server

#### How do you propose to implement this?
Check for RSIG presence then override RSSI and SNR value with it.

#### Can you do this yourself and submit a Pull Request?
@johanstokking

johanstokking, 2019-07-03:
@KrishnaIyer we should be able to do this safely for all incoming UDP traffic.

johanstokking, 2019-07-09:
We do this already

closed 2019-07-09 by johanstokking