TheThingsNetwork/lorawan-stack #933: Add channel utilization to gateway connection stats

htdvisser, 2019-07-04:
#### Summary

We would like to see channel utilization in the gateway connection stats.

Originally requested by @ajovertoom

#### Why do we need this?

It's helpful for users to see how much their gateway is used, how much each sub-band is used and how much each channel is used. 

#### What is already there? What do you see now?

We have the `uplink_count` and `downlink_count` in the API, and those fields are filled by the GS. The GS also keeps track of downlink utilization of sub-bands.

#### What is missing? What do you want to see?

To have the (downlink) utilization of a gateway's sub bands exposed through API.

#### How do you propose to implement this?

- Expose utilization from scheduling package (@johanstokking)
- Add utilization to GatewayConnectionStats API (@johanstokking)
- Fill utilization fields into GatewayConnectionStats (@johanstokking)
- Update view in Console (@bafonins or @kschiffer)

neoaggelos, 2020-05-11:
For this one, I would suggest we add a new SubBandUtilizations map[string]float32 field. For each subband, we can get the duty cycle from https://github.com/TheThingsNetwork/lorawan-stack/blob/13d2f9d6ccd7ffac56b144507e3a5ac65d464717/pkg/gatewayserver/scheduling/sub_band.go#L107.

johanstokking, 2020-05-11:
I would go with a `SubBandStatistics` struct which includes a float for duty-cycle, but potentially also other stats in the future, like SNR per DR, number of packets sent/received etc.

What would your proposal be for the `string` key? Sub-bands don't really have a name; they merely have a lower and upper frequency.

neoaggelos, 2020-05-11:
I was thinking `MinFrequency-MaxFrequency` (e.g. `863000000-865000000`) Since we are in MHz territory though, we could drop the extra zeros to make this more readable, e.g. `863-865`. Ι think that it looks OK with a single decimal too (e.g. `868.7-869.2`).

It makes more sense to go with a struct, in order to make adding more stats in the future.

johanstokking, 2020-05-12:
> to make this more readable, e.g. `863-865`. Ι think that it looks OK with a single decimal too (e.g. `868.7-869.2`).

This is not the job of the API for many reasons, including that the API doesn't know how the client will visualize this or what the decimal separator of the client is, see comments in #2513.

neoaggelos, 2020-05-12:
Reopening because we still need the Console part

johanstokking, 2020-05-13:
@neoaggelos please update issue comment

kschiffer, 2021-01-12:
Moved this to Next Up, since it is goldplating and I have too many upcoming issues already.

htdvisser, 2022-07-28:
This is now covered by the NOC.

PR #2513 (2020-05-12): Include SubBandStatistics in GatewayConnectionStats

#### Summary

Closes #933 

#### Changes

- Add new `SubBandStatistics` field in `GatewayConnectionStats` message.
- Populate that information from the Scheduler of the underlying io.Connection
- Store it with the rest of the stats

#### Notes for Reviewers

- We store uplink/downlink/status stats separately. Sub band statistics are retrieved from the scheduler, so these belong with the downlink stats.
- I am not sure if this should go on `master` or `develop`. Targetting `develop` to be on the safe side. Let's change the target to `master`  [... cut]
closed 2022-07-28 by htdvisser