TheThingsNetwork/lorawan-stack #939: Collaborator Rights Handling

htdvisser, 2019-07-04:
#### Summary

This is an overview of how collaborator rights should be handled.

#### What do you want to see?

- [ ] IS: When an Application is created, the first collaborator should have `RIGHT_APPLICATION_ALL`
- [ ] IS: When a Client is created, the first collaborator should have `RIGHT_CLIENT_ALL`
- [ ] IS: When a Gateway is created, the first collaborator should have `RIGHT_GATEWAY_ALL`
- [ ] IS: When an Organization is created, the first collaborator should have `RIGHT_ALL` ⚠
- [ ] Console: Having universal rights does **not** mean a user is admin. Use the `is_admin` bool in the response of `/api/v3/auth_info` for that.
- [ ] Console: Non-admin users can assign/revoke rights on their entities to admin users
- [ ] Console: A user can see and manage collaborators (only) if they have `RIGHT_{ENTITY}_SETTINGS_COLLABORATORS`
    - [ ] The current user can assign at most the rights that they have themselves to other users or organizations (see `/api/v3/{entity}s/{id}/rights`)
    - [ ] The current user can revoke the rights of any collaborator, regardless of their rights
    - [ ] If an existing collaborator has more rights than the current user (again based on `/api/v3/{entity}s/{id}/rights`), we can display a warning when editing that other collaborator's rights, but we should not prevent it.
    - [ ] The `RIGHT_ALL` pseudo-right implies `RIGHT_*_ALL`
    - [ ] The `RIGHT_{ENTITY}_ALL` pseudo-right implies `RIGHT_{ENTITY}_*`
    - [ ] When deselecting `RIGHT_ALL` or `RIGHT_{ENTITY}_ALL` you can expand+select the matching rights of the current user (again based on `/api/v3/{entity}s/{id}/rights`)

@kschiffer please let me know if there are other things that need to be specified.

kschiffer, 2019-08-20:
> If an existing collaborator has more rights than the current user (again based on `/api/v3/{entity}s/{id}/rights`), we can display a warning when editing that other collaborator's rights, but we should not prevent it.

One problem I have identified here is that if User A has rights that are outside of the scope of granted rights for User B, User B cannot modify the rights of User A properly because:
- Sending the out-of-scope rights together with the legitimately altered rights will result in a 403 Forbidden
- Not sending the out-of-scope together with the legitimately altered rights will not result in a 403 but will revoke these rights, even when that might not be desired

This means that in this scenario, there is no way to (legitimately) set some right while leaving out-of-scope rights intact.

How should we (or the console) handle this scenario? Currently, we prevent any right modification in such cases. We could instead notify the user of the danger: `Note: This collaborator possesses rights that are outside of the scope of your own granted rights. Modifying the collaborator's rights will also revoke these rights.`. Would that be desirable?

htdvisser, 2019-08-20:
I think it would be better to change the Access service backend and make it so that it also considers existing rights of User A when User B edits these rights. User B would be allowed to edit (add/remove) rights that User B has, but not be allowed to edit (add/remove) rights that User B does not have.

kschiffer, 2019-08-20:
Agreed. I will open an issue about this.
Until this is resolved, I think it's best to keep the right modification restricted in these cases.

htdvisser, 2019-09-11:
Closing this issue, can still be used as reference.

closed 2019-09-11 by htdvisser