TheThingsNetwork/lorawan-stack #949: Generic device format conversion

johanstokking, 2019-07-05:
#### Summary

Generic device format conversion.

#### Why do we need this?

To import devices from other sources in various formats.

#### What is already there? What do you see now?

JS `Provision()` rpc to turn provisioning data into a stream of `EndDevice` templates

#### What is missing? What do you want to see?

The idea of turning a binary file in a template of `EndDevice` that can be created on the concerning components is good, but this endpoint is not specific to Join Server. There can also be a (standardized) device exchange format with NS and AS specific entries.

#### How do you propose to implement this?

New gRPC service;
- Returns list of supported formats by the server configuration
- Converts a binary file into a stream of `EndDevice` that can either be created directly and/or be altered by the client (i.e. setting/overriding fields)

#### Can you do this yourself and submit a Pull Request?

Yes

htdvisser, 2019-07-09:
What component(s) would expose this new service? I think it would be a bit strange to expose it on the full stack (including GS), but on the other hand, we already do the same with the frequency plans RPC (expose it on the JS).

johanstokking, 2019-07-10:
This service is helpful for any component with a device registry (IS, JS, NS and AS). Some converters will only return fields for one of these components (i.e. secure element manifests typically only JS), some for multiple (i.e. import files all components).

So I'd suggest just like GCS a dedicated component that starts with IS, JS, NS and AS.

PR #1016 (2019-07-23): Device templates

#### Summary

Closes #949 
References https://github.com/TheThingsIndustries/lorawan-stack/issues/1672

#### Changes

- Add first class component Device Template Converter (DTC) that starts with anything with a device registry (IS, JS, NS, AS) or standalone
- Promote binary conversion to device templates from Join Server's provisioners to `pkg/devicetemplates`
- Add `DeviceTemplateConverter` service that converts binary input to device templates
- Extensive CLI support for working with templates
- Documentation

#### Notes for Reviewers

- Future work is making `EndDeviceTemp [... cut]
closed 2019-07-23 by johanstokking