TheThingsNetwork/lorawan-stack #960: Improve Test Coverage

htdvisser, 2019-07-09:
#### Summary

...

#### Steps to Reproduce

See Coveralls for the latest `master` branch

#### What do you see now?

The following top-50 of missed lines (excluding pkg/util/test). I already added the names of developers I think are best able to work on these.

File | Missed | Who
-- | -- | --
pkg/applicationserver/applicationserver.go | 121 | @johanstokking
pkg/applicationserver/io/web/webhooks.go | 65 | @johanstokking
pkg/band/band.go | 48 | ?
pkg/basicstation/cups/messages.go | 55 | @KrishnaIyer
pkg/basicstation/cups/server.go | 63 | @KrishnaIyer
pkg/basicstation/cups/update_info.go | 79 | @KrishnaIyer
pkg/blob/bucket.go | 54 | @htdvisser
pkg/component/component.go | 54 | ?
pkg/config/config.go | 69 | @htdvisser
pkg/config/tls.go | 44 | @htdvisser
pkg/crypto/join_messages.go | 54 | @rvolosatovs
pkg/email/registry.go | 68 | @htdvisser
pkg/email/smtp/smtp.go | 68 | @htdvisser
pkg/encoding/lorawan/lorawan.go | 252 | @rvolosatovs
pkg/encoding/lorawan/mac.go | 115 | @rvolosatovs
pkg/encoding/lorawan/messages.go | 142 | @rvolosatovs
pkg/errors/grpc.go | 60 | @htdvisser
pkg/frequencyplans/frequencyplans.go | 100 | ?
pkg/gatewayserver/gatewayserver.go | 80 | @johanstokking
pkg/gatewayserver/io/io.go | 44 | @johanstokking
pkg/gatewayserver/io/udp/udp.go | 70 | @johanstokking
pkg/identityserver/application_registry.go | 61 | @htdvisser
pkg/identityserver/client_registry.go | 63 | @htdvisser
pkg/identityserver/entity_access.go | 95 | @htdvisser
pkg/identityserver/gateway_registry.go | 63 | @htdvisser
pkg/identityserver/organization_registry.go | 50 | @htdvisser
pkg/identityserver/profile_picture.go | 48 | @htdvisser
pkg/identityserver/registry_search.go | 54 | @htdvisser
pkg/identityserver/store/store.go | 65 | @htdvisser
pkg/identityserver/user_registry.go | 151 | @htdvisser
pkg/joinserver/joinserver.go | 83 | @rvolosatovs
pkg/networkserver/downlink.go | 244 | @rvolosatovs
pkg/networkserver/grpc_deviceregistry.go | 150 | @rvolosatovs
pkg/networkserver/grpc_gsns.go | 216 | @rvolosatovs
pkg/networkserver/mac_link_adr.go | 62 | @rvolosatovs
pkg/networkserver/utils.go | 86 | @rvolosatovs
pkg/oauth/storage.go | 57 | @htdvisser
pkg/pfconfig/shared/shared.go | 88 | @KrishnaIyer
pkg/redis/redis.go | 98 | @rvolosatovs
pkg/ttnpb/contact_info.go | 50 | @htdvisser
pkg/ttnpb/identifiers.go | 45 | @htdvisser
pkg/ttnpb/lorawan_mac_commands.go | 251 | @rvolosatovs
pkg/ttnpb/lorawan_populate.go | 286 | @rvolosatovs
pkg/ttnpb/lorawan.go | 462 | @rvolosatovs
pkg/ttnpb/messages.go | 45 | ?
pkg/types/devaddr.go | 45 | ?

There are also still files with zero coverage:

File | Covered | Who
-- | -- | --
pkg/applicationserver/io/formatters/protobuf.go | 0 | @johanstokking
pkg/auth/pkix.go | 0 | @johanstokking
pkg/band/downgrades.go | 0 | ?
pkg/component/configuration_grpc.go | 0 | @htdvisser
pkg/config/shared.go | 0 | @htdvisser
pkg/config/tls.go | 0 | @htdvisser
pkg/email/message.go | 0 | @htdvisser
pkg/email/registry.go | 0 | @htdvisser
pkg/email/smtp/smtp.go | 0 | @htdvisser
pkg/events/correlation_interceptor.go | 0 | @htdvisser
pkg/identityserver/profile_picture.go | 0 | @htdvisser
pkg/joinserver/grpc_asjs.go | 0 | @rvolosatovs
pkg/joinserver/grpc_nsjs.go | 0 | @rvolosatovs
pkg/log/global.go | 0 | @htdvisser
pkg/log/noop.go | 0 | @htdvisser
pkg/oauth/callback.go | 0 | @htdvisser
pkg/ttnpb/application.go | 0 | @rvolosatovs
pkg/ttnpb/applicationserver_web.go | 0 | @johanstokking
pkg/ttnpb/applicationserver.go | 0 | @johanstokking
pkg/ttnpb/contact_info.go | 0 | @htdvisser
pkg/ttnpb/end_device.go | 0 | @rvolosatovs
pkg/ttnpb/gateway.go | 0 | @htdvisser
pkg/ttnpb/join_populate.go | 0 | @rvolosatovs
pkg/ttnpb/keys.go | 0 | @htdvisser
pkg/ttnpb/lorawan_mac_commands.go | 0 | @rvolosatovs
pkg/ttnpb/message_validators.go | 0 | @rvolosatovs
pkg/ttnpb/messages.go | 0 | ?
pkg/ttnpb/organization.go | 0 | @htdvisser
pkg/ttnpb/user.go | 0 | @htdvisser
pkg/web/logger.go | 0 | ?
pkg/web/middleware/recover.go | 0 | ?
pkg/web/redirect.go | 0 | ?

#### What do you want to see instead?

- We should try to cover relevant lines with realistic tests.
- I think we should ignore coverage of `pkg/util/test` in Coveralls so that it doesn't affect our coverage percentage.
- In the future we could also consider ignoring things like `if err != nil { return err }` in coverage.

#### How do you propose to implement this?

When you're blocked or waiting for review of a feature, we should try to spend some time on improving tests. 

#### Can you do this yourself and submit a Pull Request?

Everyone just work on their own packages. Help from others is welcome.

rvolosatovs, 2019-07-09:
Refs #918

johanstokking, 2019-07-09:
@TheThingsNetwork/stack-developers please assign low hanging fruit to @furtiman @mjamescompton to get acquainted with the code base and testing processes

johanstokking, 2019-09-11:
@htdvisser please update table

The goal here is 75% coverage on the package-level, the relevant parts of the packages. Best is to open the package you're assigned to and view test coverage in your editor.

We can discuss individual exceptions here.

htdvisser, 2019-09-20:
I think Coveralls is pretty clear here: https://coveralls.io/github/TheThingsNetwork/lorawan-stack

[image]

Some of these packages (blob, email) require external services (s3,gcs,smtp,...) to be configured, and some of those tests are currently skipped by Travis, so we can ignore those

I think we should aim for more than 75% if possible, but of course we shouldn't spend hours on tests that cover each and every line. In many cases we can accept that lines like these aren't covered:

```go
if err != nil {
  return _, err
}
```

(it would be nice if we could filter those kind of lines out of coverage reports so that we'd know what really requires work)

htdvisser, 2022-06-29:
The coverage table in this issue is quite outdated. And with the Coveralls integration gone, we can't regenerate it. I think we should just close the issue and generate a new overview at some point in the future.

closed 2022-06-29 by htdvisser