TheThingsNetwork/lorawan-stack #970: Release from release branches

johanstokking, 2019-07-10:
#### Summary

Release from release branches.

#### Why do we need this?

To avoid triggering releases from anyone pushing tags accidentally.

#### What is already there? What do you see now?

Trigger releases on pushed tags in `master` branch.

#### What is missing? What do you want to see?

Trigger releases on pushed tags in `release/v*` branches.

#### How do you propose to implement this?

- [x] Setup branch protection for release branches on GitHub
- [ ] Apply release rules for Travis on release branches (i.e. `branch =~ /^release\/v.*/`) instead of `master`, still requiring a tag. So we want CI checks for both `master` _and_ release branches, and only release from release branches

#### Can you do this yourself and submit a Pull Request?

Will review

adriansmares, 2019-07-10:
Should we clean dependency caches for release builds as well ?
https://github.com/TheThingsNetwork/lorawan-stack/blob/4163efd780040e5d04b35fc7d6c68d443258ebd9/.travis.yml#L37-L43

htdvisser, 2019-07-10:
As already discussed on Slack, tags aren't pushed on branches, so you can't _"Trigger releases on pushed tags in `release/v*` branches"_.

If the goal is to only trigger releases on actual release tags (and not other tags), we can make Travis check that the release tags are the same as we'd expect from a release tag generated by our Mage tooling:

```
% git tag -v $TRAVIS_TAG 2> /dev/null | grep -E '^tagger'
tagger Johan Stokking <johan@thethingsnetwork.org> 1557491575 +0200
```

```
% git tag -v $TRAVIS_TAG 2> /dev/null | grep -E '^(Pre-r|R)elease'
Release version 3.0.3
```

johanstokking, 2019-09-11:
@adriansmares stiped out the second checkbox; this doesn't work when combining Travis and GoReleaser apparently. So it's just a small clarification in Travis to regex the tag name.

PR #1390 (2019-10-01): Release only on tagged releases

#### Summary

Closes #970

#### Changes

- Do `goreleaser` releases only if the tag respects the `/^v3\.[0-9]+\.[0-9]+/` regex.

#### Notes for Reviewers

We allow the `release` build to run even if we don't have a correct tag, in order to allow pre-releases (that `goreleaser` will release as snapshots).
closed 2019-10-01 by adriansmares