TheThingsNetwork/lorawan-stack #971: Password Reset from web UI

htdvisser, 2019-07-10:
#### Summary

Users should be able to reset their password using the web UI.

Old issue: https://github.com/TheThingsIndustries/lorawan-stack/issues/1123

#### Why do we need this?

Because not all users are comfortable with the API or CLI

#### What is already there? What do you see now?

We have the `CreateTemporaryPassword` RPC in the `UserRegistry` service. HTTP binding is `POST /api/v3/users/{user_ids.user_id}/temporary_password`. This needs to be called with the user ID and Email Address. A successful call results in an email being sent to the user with a temporary password that the can only use to reset their password with the `UpdatePassword` RPC in the `UserRegistry` service.

#### What is missing? What do you want to see?

The following screens in the Web UI of the Identity Server:

- A "Forgot Password" screen where the user has to enter their username and email in order to receive a temporary password.
- A "Reset Password" screen where the user can enter their username, their (current or temporary) password and a new password (+confirmation). In the future this screen will also get a checkbox that will make the IS revoke all sessions and access tokens #267.

#### How do you propose to implement this?

Assigning @pgalic96 since he was already assigned to the original issue. Please discuss with @kschiffer the design. I propose to implement it in a (new) "account" React app that will also have the screen I discussed in #265. Note that the user does not need to be logged in for the screens proposed in this issue. Also note that these must **not** be implemented in the Console, since the Console must never have access to the user's password.

pgalic96, 2019-07-18:
> A "Forgot Password" screen where the user has to enter their username and email in order to receive a temporary password.

Should it be username and email or just e-mail?

cc @kschiffer

johanstokking, 2019-07-18:
I think that people are better at remembering their email address than their user name, so please file an issue to be able to (only) specify their email address.

kschiffer, 2019-07-18:
> I propose to implement it in a (new) "account" React app that will also have the screen I discussed in #265.

Isn't the OAuth react app already the "account" app you're describing here, @htdvisser?

PR #1042 (2019-08-14): Reset and change password UI

#### Summary

Closes #971

#### Changes

- Add `reset password` view for sending a reset password request
- Add `change password` view for resetting password after temporary token is obtained
- Add api requests for reset and change password

#### Notes for Reviewers

- Please note that the `change password` view is not intended to be accessed with a button, at least not for the user that is not logged in, we can reuse this view to make a change password view for the logged in user, so not in case of password reset
- when sending a reset password request, in the st [... cut]
closed 2019-08-14 by pgalic96