TheThingsNetwork/lorawan-stack #975: Avoid unhandled promise rejections in the request logics

kschiffer, 2019-07-10:
#### Summary
With the way our request promise middleware works at the moment, it will produce unhandled promise rejections, when a request fails and the dispatch is not handled via a try/catch block.

#### Steps to Reproduce
1. Login to the console
2. Initiate an action that will fail (as in produce a `{BASE_ACTION}_FAILURE` action type)
3. Notice unhandled promise rejections in the logs

#### What do you see now?
Unhandled promise rejections for dispatches of request actions, when the dispatch is not wrapped in try/catch and the dispatch results in a failure action.

#### What do you want to see instead?
No unhandled promise rejections happening on failure actions

#### Environment
Console.

#### How do you propose to implement this?
The issue is caused [by this block](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/store/middleware/logics/lib.js#L65-L68).
This can be fixed by resolving the promise on request failures as well. As far as the logic is concerned, this is ok because the logic still ran through as expected. We should reject the promise only when the dispatch or action creators itself threw an error.

Using this fix, failed request action will not cause the promise returned by the dispatch to reject, which solves the issue. In any case, determining whether an action resulted in success or failure is to be done by examining the error values in the store following the result action of the request.

#### Can you do this yourself and submit a Pull Request?
Yes.

kschiffer, 2019-07-10:
So after investigating this for a little, I can see the following approaches to fix this:

- Don't reject the promise at all, work with the resolved value instead
  - We would need to replace the `try/catch` blocks around dispatches (mostly in form submits)
  - We could resolve the promise with the result action of the request action, and then probe for the action's `error` prop, which we can assume to be present due to complying with Flux Standard Action
  - See #979 for this approach
- Remove the promise middleware altogether and move the api request logic from the store logic to the form submit handlers
  - We would then need to call the update success actions manually
- Remove the promise middleware and work with `fetching` and `error` store values
  - To enable something like toasts then would be quite fiddly because we'd need to react on the `fetching` value change
  - If we pass update action error messages from the store to the components, they would persist through unmounts. This means we would need to apply actions to clear the error values after unmounting
- Ignore unhandled promise rejections, or handle them via event handlers
  - Not desirable IMO

bafonins, 2019-07-15:
cant reproduce the issue

kschiffer, 2019-07-17:
You will get the issue for example on the gateway page, when the gateway is not connected:
```js
Uncaught (in promise) {code: 5, message: "error:pkg/gatewayserver:not_connected (gateway `my-awesome-gateway` not connected)", details: Array(1)}
```
The `UPDATE_GATEWAY_STATISTICS_REQUEST` will fail and make the promise middleware [`reject()` the promise](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/store/middleware/logics/lib.js#L65-L68), leading to an unhandled promise rejection, because there is a promise returned by the dispatch that is not handled anywhere.

bafonins, 2019-07-23:
The problem here is that we treat all `_REQUEST` actions the same way. 
Actions triggered by form submission are different from, say, getting a list of entities. In the console/oauth forms are absolutely self-contained:
1. The form keeps its `submitting` value in the react state
2. There is no reason to store errors that happen on form submission in the redux state
3. We want to `await` only on form submission to perform ui changes like toasts/redirects/etc

Instead of altering the promise middleware altogether, I propose to change action creators that are used by forms across the app. This could look something like this:
```
  UPDATE_GATEWAY_REQUEST_FORM
  UPDATE_GATEWAY_SUCCESS_FORM
  UPDATE_GATEWAY_FAILURE_FORM
```

and in the promise middleware we would check for `_REQUEST_FORM` action types.

kschiffer, 2019-07-23:
> The form keeps its the submitting value in the react state
There is no reason to store errors that happen on form submission in the redux state
We want to await only on form submission to perform ui changes like toasts/redirects/etc

Indeed.

>   UPDATE_GATEWAY_REQUEST_FORM
  UPDATE_GATEWAY_SUCCESS_FORM
  UPDATE_GATEWAY_FAILURE_FORM

What I don't like about this is that we would attach yet more implicit logic based on the action type. We might also end up having different action types for the same logic, the only difference being whether a promise is attached or not.

An alternative could be to use a special wrapper around action creators that will decorate the action with a flag inside `meta`, which will be read by the promise middleware:

```js
dispatch(attachPromise(updateEntity(patch)))
```
The advantage I see here is that we would have more control over when a promise is attached to an action, better ensuring that we handle it properly whenever we use it. We could then choose to attach promises to any request action we want and avoid implicit side-effects by logic attached to action type values.

bafonins, 2019-07-24:
sure

PR #1079 (2019-08-01): Fix unhandled promise rejections of action dispatches
#### Summary
Closes #975.

#### Changes
- Change promise attacher middleware to check for an `_attachPromise` flag to be true
- Change `createRequestLogic` to resolve/reject only when a promise is attached
- Add a wrapping function for action creators that will attach the `_attachPromise` flag to the `meta`-object of the action
- Refactor awaited dispatches to use the new action creator wrapper

#### Notes for Reviewers
This is the solution according to our discussion in #975

#### Release Notes
- Fixed an issue resulting in errors being un [... cut]
closed 2019-08-01 by kschiffer