TheThingsNetwork/lorawan-stack #977: Valid web paths return 404

adriansmares, 2019-07-10:
#### Summary

#919 introduced a regression related to web paths - it seems that one of the dependency updates broke down the route handling. 
@kschiffer observed this while attempting to `GET` an application link (`GET http://localhost:1885/api/v3/as/applications/app1/link?field_mask=api_key,network_server_address`)

#### Steps to Reproduce

1. Create an application `app1`
2. Set the application link for `app1` through the CLI.
3. Attempt to `GET http://localhost:1885/api/v3/as/applications/app1/link?field_mask=api_key,network_server_address`

#### What do you see now?

```
"error:pkg/errors/web:http (HTTP error: code=404, message=Not Found)"
```

#### What do you want to see instead?

```
{
  "code": 7,
  "message": "error:pkg/auth/rights:no_application_rights (no rights for application `app1`)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/auth/rights",
      "name": "no_application_rights",
      "message_format": "no rights for application `{uid}`",
      "attributes": {
        "uid": "app1"
      },
      "correlation_id": "63d6d3b76c4a444ba939fc6b7e5da015"
    }
  ]
}
```

#### Environment

On d6f3b8df290d6741bdd1a81da321698cb41f9fab (the commit that merged #919) the resulting output is the `404`. On 6fe2c82241ad2cfc7a610c8b5ef9071d4a3fcef1 (the `master` before the merge) the result is the `no_application_rights`.

#### How do you propose to implement this?

No idea yet.

#### Can you do this yourself and submit a Pull Request?

I'll continue to investigate which package introduced this regression and report it here.

htdvisser, 2019-07-10:
The tests were actually wrong before that commit. Creating an Echo group is and should be a no-op until you actually register routes/handlers to it.

adriansmares, 2019-07-10:
I'm not really sure it is a real no-op:
https://github.com/labstack/echo/blob/master/echo.go#L531-L535
https://github.com/labstack/echo/blob/master/group.go#L21-L30
If the group contains has a default middleware, a default route to the not found handler is created when you actually create the group.

adriansmares, 2019-07-10:
And this default handler, links to our issue at hand.
https://github.com/TheThingsNetwork/lorawan-stack/blob/6d12c1e25dab0f36698bc1209bab4c86d3eb3fd8/pkg/component/grpc.go#L58
Before Echo `v4.1.3`, this `NotFoundHandler` would have been registered on the actual root group of the Echo server, but in `v4.1.3` they changed it to just be registered only inside the group (see [this](https://github.com/labstack/echo/compare/v4.1.2..v4.1.3#diff-103d48dbdf9e16c6eb09b90668a9e984R26)).
This results in a conflict - we have our own handler for `/*` that passes the request to gRPC, but since the group creation also creates this handler for `/*`, and both of them exist in the same group, the not found handler has a higher priority by being first. This wasn't an issue before `v4.1.3` since groups have higher priority than root paths.

I'm going to report this to Echo, but in the meantime if we want to restore master back to normal functioning, I can submit a PR that downgrades Echo to `v4.1.2` and we create an issue to remind us that at some point we have to upgrade.

johanstokking, 2019-07-10:
Looks like they shouldn't be doing this stuff in patches indeed.

adriansmares, 2019-07-11:
References https://github.com/labstack/echo/issues/1365

PR #985 (2019-07-11): dev: Downgrade Echo to v4.1.2

#### Summary

Closes #977

#### Changes

- Downgrade Echo from v4.1.2 to v4.1.6

#### Notes for Reviewers

None.
closed 2019-07-11 by adriansmares