TheThingsNetwork/lorawan-stack #996: Hide "start event stream" event

johanstokking, 2019-07-15:
#### Summary

Hide "start event stream" event

cc @janjongboom

#### Why do we need this?

To not emit events that come from user actions giving the impression as something happened.

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

Nothing. This is no activity 

#### How do you propose to implement this?

Remove start events

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2019-07-15:
If you remove the stream start/stop events, make sure to explicitly send the header from the RPC implementation, otherwise the request may time out on the client side.

```diff
diff --git a/pkg/events/grpc/grpc.go b/pkg/events/grpc/grpc.go
index 0bd0897f5..7e30902c9 100644
--- a/pkg/events/grpc/grpc.go
+++ b/pkg/events/grpc/grpc.go
@@ -26,6 +26,7 @@ import (
 	"go.thethings.network/lorawan-stack/pkg/rpcmiddleware/warning"
 	"go.thethings.network/lorawan-stack/pkg/ttnpb"
 	"google.golang.org/grpc"
+	"google.golang.org/grpc/metadata"
 )
 
 const workersPerCPU = 2
@@ -78,11 +79,6 @@ type EventsServer struct {
 	filter events.IdentifierFilter
 }
 
-var (
-	evtStreamStart = events.Define("events.stream.start", "start event stream")
-	evtStreamStop  = events.Define("events.stream.stop", "stop event stream")
-)
-
 // Stream implements the EventsServer interface.
 func (srv *EventsServer) Stream(req *ttnpb.StreamEventsRequest, stream ttnpb.Events_StreamServer) (err error) {
 	ctx := stream.Context()
@@ -120,8 +116,9 @@ func (srv *EventsServer) Stream(req *ttnpb.StreamEventsRequest, stream ttnpb.Eve
 		warning.Add(ctx, "Historical events not implemented")
 	}
 
-	srv.pubsub.Publish(evtStreamStart(ctx, req, req))
-	defer srv.pubsub.Publish(evtStreamStop(ctx, req, req))
+	if err = stream.SendHeader(metadata.MD{}); err != nil {
+		return err
+	}
 
 	for {
 		select {

```

johanstokking, 2019-07-15:
I'm good with emitting the event from the server but this is mostly about not showing it in the Console, but just an indicator that event streaming is connected (based on that event)

htdvisser, 2019-07-15:
That would also be a bit complicated. If you have the same screen open in multiple tabs, then you'll have multiple streams and therefore multiple start/stop events. In order to determine which event marks the start/stop of the event stream of the current tab, the console would have to compare the event correlation IDs to the request ID in the response header.

johanstokking, 2019-07-15:
How about just hiding the start/stop and showing a green indicator if any event (like start) has been received? We also know when the stream stops, right?

Console A really doesn't have to show anything about event streams on Console B

kschiffer, 2019-07-16:
We already have a (blue) indicator when the event stream was opened successfully. These are not triggered by the start event, but simply by having successfully opened the streaming endpoint. Would that be enough as indication?

Also, would this mean that we should hide events from user edits, like a user editing the name of a gateway. Currently this will trigger an event that will be shown in the event view.

htdvisser, 2019-07-16:
Didn't we discuss that we'd add event type filters in the future? When we implement that, we can just define a default filter set for different places where we show an event component.

johanstokking, 2019-07-16:
The filter makes sense. Still, the event start/stop can be filtered out anyway.

bafonins, 2020-10-03:
Already implemented. Also see https://github.com/TheThingsNetwork/lorawan-stack/pull/2989

closed 2020-10-03 by bafonins