TheThingsNetwork/lorawan-stack #997: Increase clickable area

johanstokking, 2019-07-15:
#### Summary

Increase clickable area

cc @janjongboom

#### Why do we need this?

Guide users

#### What is already there? What do you see now?

Small margins on primary buttons

#### What is missing? What do you want to see?

Higher margins, bigger buttons

Potentially also a color shade for other actions/buttons that are not primary, i.e. copy EUI/key buttons, navigation, breadcrumbs, etc

#### How do you propose to implement this?

Probably some CSS

#### Can you do this yourself and submit a Pull Request?

Can review

kschiffer, 2019-07-17:
I disagree here. The margins/paddings we are using on buttons are not any smaller than, say, on GitHub. For users that would like bigger dimensions, there is always the option to zoom via the browser. 
Also, we have primary and secondary (and even more) button and link colors, which we use consistently (e.g. secondary button colors for EUI/key copy buttons).

johanstokking, 2019-07-17:
Imho the difference is quite big;

GitHub:

[image]

V3 Console:

[image]

It's probably a combination of font, font size, font weight and padding, but the GitHub buttons are definitely more friendly to click on.

kschiffer, 2019-07-17:
In terms of height, they're exactly the same. Only the left/right paddings are smaller which I can adjust. Font size and weight might need some adjustments towards better legibility, but they don't influence the clickable area.

johanstokking, 2019-11-12:
We don't consider this a problem

closed 2019-11-12 by johanstokking