TheThingsNetwork/lorawan-stack #1005: Custom variable support in Webhook URLs

adriansmares, 2019-07-16:
#### Summary

Currently the final URL of a webhook is static, it does not support custom variables such as `JoinEUI`, `DevID`, `DevEUI` and `DevAddr`. The result should be that a base URL for webhook such as `http://example.com/app1/wh1` can become `http://example.com/app1/wh1/{{.DevEUI}}` and allow the Webhook endpoint to route the uplinks better. This should also be applicable inside the message specific URLs (uplink message, downlink queued etc.)

#### Why do we need this?

This will allow better routing of the Webhooks and also enable us to write certain integrations (such as OpenSensors) only as WebHooks.

#### What is already there? What do you see now?

Static URL support - `BaseURL` is joined with the topic based on the uplink type, but there are no custom routing fields.

#### What is missing? What do you want to see?

The custom routing fields `{{.AppEUI}}`, `{{.JoinEUI}}`, `{{.DevID}}`, `{{.DevEUI}}`, `{{.DevAddr}}`.

#### Environment

Not applicable.

#### How do you propose to implement this?

Replace custom variables in the URL after the base URL + config URL join occurs in `newRequest.`

#### Can you do this yourself and submit a Pull Request?

Yes.

htdvisser, 2019-07-16:
Based on how the fields look, I suspect that you were planning on implementing this with `text/template`. If that's indeed the case, then please be aware of the fact that parsing **and** executing templates (especially if it's not just the base URL but also the path) on every uplink can potentially be quite resource intensive.
Also, make sure to use a very limited struct that the template will be executed with. Use only primitive types (so convert all EUIs/DevAddr to strings), avoid pointers.

adriansmares, 2019-07-16:
I think we can cache the template itself using the Webhook ID + `updated_at` in order to avoid recompiling the template every time. Also it is indeed the case that only strings should be passed to the template itself (and I'd prefer we don't use anything else outside of the identifiers).

johanstokking, 2019-07-16:
I think two options;

- Use `text/template` with a very simple struct specific for this use case. I.e., don't pass `*ttnpb.ApplicationUp` but `WebhookTemplateFields` with only the stuff that we support
- Use string replace on the known fields. Super fast, same result

htdvisser, 2019-07-16:
If we go for the second, then we should probably not use the Go template syntax at all, because that would just confuse people (why the dot? is it padded with spaces?).

If we want to keep the validation rules the same, we could consider using placeholders like `$app_id`. That way the placeholders won't be valid in the host (because we don't want that), but they will be valid in the path (I think `$` is allowed in URL paths, we need to check that).

johanstokking, 2019-07-17:
What do you mean with "allow one variable"?

I also think that user experience-wise, it's nicer not to use `text/template`, apart from the performance hit. I'm fine with PHP-style variable names.

adriansmares, 2019-07-17:
The standard mentioned above would allow expressions such as `{appEUI,devEUI}`, which would not allow us to use the simple string replace. Note that I did not disallow expressions such as `{appEUI}/{devEUI}`, which have only one variable per capture.

johanstokking, 2019-07-17:
[How about this one?](https://godoc.org/google.golang.org/api/googleapi/internal/uritemplates)

adriansmares, 2019-07-17:
That one is an internal package and cannot be imported outside of the main package, but [this](https://github.com/jtacoma/uritemplates) should do it.

PR #1011 (2019-07-24): Webhooks template support and path variables

#### Summary

Closes #1005 
References #945

#### Changes

- Add support for template variables inside the Webhook URLs. Currently supported variables are `appID`, `appEUI`, `joinEUI`, `devID`, `devEUI` and `devAddr`.
- Add Webhook templates support in `pkg/web`. The templates are defined as protos and are based on fields, which are service specific fields that differ from integration from integration.

#### Notes for Reviewers

This PR adds the initial groundwork for #945, by adding the structure of the webhook templates and their storage RPCs. Th [... cut]
closed 2019-07-24 by adriansmares