TheThingsNetwork/lorawan-stack #1018: Show not found view on non-existing routes

bafonins, 2019-07-19:
#### Summary

We should inform the user when visiting non-existing routes of application/gateway sub views.
Currently, we handle only the root routes, i.e. `/console/{route}`

#### Steps to Reproduce

1. Go to `/console/applications/{existing-id}/does-not-exist
2. See empty page

#### What do you see now?

[image]

#### What do you want to see instead?

Something like this:
[image]

#### Environment

browser (console)

#### How do you propose to implement this?

TBA

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2019-09-17:
Not sure exactly how to integrate it into error handling logic we have. I see two options:

Just throw an error on unknown route:
```
lib/components/not-found-route

export default () => {
  throw { statusCode: 404 }
}
```

```diff
// views/application
export default class Application extends React.Component {
  componentWillUnmount() {
    const { appId, stopStream } = this.props

    stopStream(appId)
  }

  render() {
    const { match, application, appId, env } = this.props

    return (
      <BreadcrumbView>
        <IntlHelmet titleTemplate={`%s - ${application.name || appId} - ${env.siteName}`} />
        <Switch>
          <Route exact path={`${match.path}`} component={ApplicationOverview} />
          <Route path={`${match.path}/general-settings`} component={ApplicationGeneralSettings} />
          <Route path={`${match.path}/api-keys`} component={ApplicationApiKeys} />
          <Route path={`${match.path}/link`} component={ApplicationLink} />
          <Route path={`${match.path}/devices`} component={Devices} />
          <Route path={`${match.path}/collaborators`} component={ApplicationCollaborators} />
          <Route path={`${match.path}/data`} component={ApplicationData} />
          <Route
            path={`${match.path}/payload-formatters`}
            component={ApplicationPayloadFormatters}
          />
          <Route path={`${match.path}/integrations`} component={ApplicationIntegrations} />
+         <NotFoundRoute />
        </Switch>
      </BreadcrumbView>
    )
  }
}
```

or use `react-router`:

```
const NotFoundPage = () => <FullViewErrorInner error={{ statusCode: 404 }} />;

const NotFoundRoute = () => <Redirect to={{ state: { notFoundPage: true } }} />;

const NotFoundPageHandler = withRouter(({children, location}) => {
  return location && location.state && location.state.notFoundPage ? <NotFoundPage /> : children;
});
```

```diff
// views/landing
export default class Landing extends React.PureComponent {
  render() {
    return (
      <div className={style.container}>
+ 		<NotFoundPageHandler>
	        <ToastContainer />
	        <WithAuth>
	          <Switch>
	            <Route exact path="/" component={Overview} />
	            <Route path="/applications" component={Applications} />
	            <Route path="/gateways" component={Gateways} />
	            <Route path="/organizations" component={Organizations} />
	            <Route component={GenericNotFound} />
	          </Switch>
	        </WithAuth>
+   	</NotFoundPageHandler>
      </div>
    )
  }
}

// views/application
export default class Application extends React.Component {
  componentWillUnmount() {
    const { appId, stopStream } = this.props

    stopStream(appId)
  }

  render() {
    const { match, application, appId, env } = this.props

    return (
      <BreadcrumbView>
        <IntlHelmet titleTemplate={`%s - ${application.name || appId} - ${env.siteName}`} />
        <Switch>
          <Route exact path={`${match.path}`} component={ApplicationOverview} />
          <Route path={`${match.path}/general-settings`} component={ApplicationGeneralSettings} />
          <Route path={`${match.path}/api-keys`} component={ApplicationApiKeys} />
          <Route path={`${match.path}/link`} component={ApplicationLink} />
          <Route path={`${match.path}/devices`} component={Devices} />
          <Route path={`${match.path}/collaborators`} component={ApplicationCollaborators} />
          <Route path={`${match.path}/data`} component={ApplicationData} />
          <Route
            path={`${match.path}/payload-formatters`}
            component={ApplicationPayloadFormatters}
          />
          <Route path={`${match.path}/integrations`} component={ApplicationIntegrations} />
+         <NotFoundRoute />
        </Switch>
      </BreadcrumbView>
    )
  }
}
```

cc @kschiffer

kschiffer, 2019-10-04:
The second option seems superior to me, as it solves the problem once at the top level. Do you have any concerns to use it?

bafonins, 2019-10-09:
@kschiffer the second solution overrides the url, not sure if it is a problem

kschiffer, 2019-10-09:
Ah right. I think we should avoid that as it would be unconventional.

This one would:
- Retain the URL
- Keep routing via `react-router` intact
- Use our error boundaries
- Use the full view error (not subview)

Right?

bafonins, 2019-10-09:
Correct

closed 2019-10-10 by bafonins