TheThingsNetwork/lorawan-stack #1029: Getting started dead links and heading refs

johanstokking, 2019-07-22:
#### Summary

Dead links in Getting Started and unnecessary heading references.

#### Steps to Reproduce

1. Run `mage docs:server`
2. Go to http://localhost:1313/gettingstarted/configuration/

#### What do you see now?

Issues:

- Dead link; "configuration documentation"
- Dead link; "networking documentation"
- Potentially other dead links

Heading references (`a name` tags)

#### What do you want to see instead?

No dead links and no named headers.

#### How do you propose to implement this?

Clean up docs

#### Can you do this yourself and submit a Pull Request?

Can review

htdvisser, 2019-07-22:
Let's make sure to always use Hugo's `relref` shortcode for internal references. Those are checked, and we'll see errors/warnings if they break.

johanstokking, 2019-07-23:
Taking this one over

PR #1040 (2019-07-23): Documentation structure and first guides

#### Summary

Closes #401 
Closes #953 
Closes #1029 

Blocks #206
Blocks #598
Blocks #757 
Blocks #1016

#### Changes

- Reorganize to primary sections: Guides, Concepts, Components and Reference
- Move Getting Started to Guides
- Merge CLI part of Getting Started in single page
- Extract and guides for Events, Multicast and Class C, Downlink Queue Operations
- Rename docs to "TTN Stack for LoRaWAN" for SEO

#### Notes for Reviewers

Please review locally:

```bash
$ mage docs:server
```

And go to http://localhost:1313

#### Release Notes
closed 2019-07-23 by johanstokking