TheThingsNetwork/lorawan-stack #1030: User can open create account page while logged in

pgalic96, 2019-07-22:
#### Summary

If user is logged in, they can still open the create account page and create a new account, which results in a newly created account being the one that is logged in. This is not intended I believe and leads to confusion.

#### Steps to Reproduce

1. Log into `oauth` client
2. Go to url `/oauth/register`
3. Create account

#### What do you see now?

Create account form

#### What do you want to see instead?

Redirect to for example, `/oauth`.

#### Environment

Latest master

#### How do you propose to implement this?

Add check if the user is already logged in and redirect

#### Can you do this yourself and submit a Pull Request?

I can do it

cc @kschiffer @bafonins

htdvisser, 2019-07-22:
It is not actually forbidden to create user accounts while you're already logged in yourself. Imagine an admin account creating user accounts for other users. Not sure if that should be done through the `/oauth/register` page. 

(TBH I think the existing registration page doesn't really belong under `/oauth` either)

kschiffer, 2019-07-22:
> It is not actually forbidden to create user accounts while you're already logged in yourself. Imagine an admin account creating user accounts for other users.

I get that, but creating users through the `Register` view is not the right UX for this. We will need a proper section somewhere in an admin config view for this. So I would recommend preventing to display the registration view when the user is already logged in.

> (TBH I think the existing registration page doesn't really belong under /oauth either)

We need an issue to discuss solutions here. I will follow up on this.

pgalic96, 2019-07-22:
@htdvisser that's why I specified this from the user interface point of view, I get that this is a desired behavior but we should create some kind of a different admin UI for it in console/oauth app

PR #1063 (2019-08-05): Redirect logged in user when trying to access Create Account page

#### Summary

- closes #1030 

#### Changes

- Add check if the user object is present and if it is return `Redirect` component that returns to `/oauth`
closed 2019-08-05 by pgalic96