TheThingsNetwork/lorawan-stack #1033: Adding higher max-eirp to custom band plan does not increase tx_power

ama9910, 2019-07-22:
#### Summary
When creating a custom band plan and adding a higher max-eirp, network does not increase tx_power sent to gateway.

#### Steps to Reproduce

- Start stack with config option which will load band plans from a custom folder
- Create custom band plan, based on AS_920_923 and add max-eirp:27
- Register device for band plan
- Register gateway for band plan
- Monitor gateway output power with ./ttn-lw-cli events --gateway-id your-gateway | jq .data.scheduled.downlink.tx_power 
- Send downlinks to device
- Note that tx_power will be 16dBm (default for plan)
- Move device so that uplink SNR is around -15dBm
- Note that tx_power is still at 16dBm

#### What do you see now?
Gateway is using 16dBm power

#### What do you want to see instead?
Gateway uses power level up to max-eirp setting of 27dBm

Not sure where the problem is, or if I misunderstand how this works.

#### Can you do this yourself and submit a Pull Request?
Unfortunately not.

johanstokking, 2019-07-22:
Thanks for reporting, this seems like a valid issue.

The concerning code is here;

https://github.com/TheThingsNetwork/lorawan-stack/blob/d1f73432f11308970a6b4af87abe88eb6b682f29/pkg/gatewayserver/io/io.go#L288-L294

So currently, the frequency's maximum EIRP may not exceed the band's; it's can only be smaller. We should consider taking the frequency plan's maximum EIRP if it's set and if it's not configured for the sub-band. Because the AS923 default EIRP is clearly a default:

>1165 AS923MHz ISM band end-devices should use the following default parameters
>1166  - Default EIRP: 16 dBm

johanstokking, 2019-07-22:
@KrishnaIyer can you take this this month?

KrishnaIyer, 2019-07-22:
Yes will do.

PR #1043 (2019-07-23): Take frequency plan's max EIRP as override when set

#### Summary

Closes #1033 
References #300 

#### Changes

- Take the frequency plan's max EIRP as override when set. This changes behavior and is documented in https://github.com/TheThingsNetwork/lorawan-frequency-plans already

#### Release Notes

- Changed interpretation of frequency plan's maximum EIRP from a ceiling to a overriding value of any band (PHY) settings
closed 2019-07-23 by johanstokking