TheThingsNetwork/lorawan-stack #1067: Make password rules configurable

pgalic96, 2019-07-29:
#### Summary

We should have a config file that would define the password rules so that they are not hardcoded in the front-end.

#### Why do we need this?

Currently password rules are defined in oauth front-end as `const`. If we change them such that they are config, we can change and extend them independently of the codebase.

#### What is already there? What do you see now?

Password rules are hard-coded in every file where password validation is used.

#### What is missing? What do you want to see?

Config file where all password rules would be defined.

#### Environment

Current master

#### How do you propose to implement this?

Have a config file and include it wherever we need validation.

#### Can you do this yourself and submit a Pull Request?

Yes, or someone else front-end related.

cc @kschiffer

johanstokking, 2019-08-07:
Please use the IS configured rules; https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/identityserver/identityserver.go#L54-L60

johanstokking, 2019-08-07:
@pgalic96 _always_ set a milestone

johanstokking, 2019-08-07:
cc @bafonins

bafonins, 2019-08-27:
@htdvisser Shall we merge [PasswordRequirements](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/identityserver/identityserver.go#L54-L60) with [OAuth Config](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/oauth/server.go#L76-L79) so the oauth server has access to those values and can embed them into `APP_DATA`?

bafonins, 2019-08-28:
Putting this one on hold until we decide how to properly propagate the is config to the oauth app

jpmeijers, 2020-02-15:
I was a little baffled by the default password complexity rule:
[image]

I thought that the modern default complexity rules should be on length, not on special characters as those don't make a password stronger.

See the NIST guidelines, Appendix A:
https://pages.nist.gov/800-63-3/sp800-63b.html#appA

> Research has shown, however, that users respond in very predictable ways to the requirements imposed by composition rules [Policies]. For example, a user that might have chosen “password” as their password would be relatively likely to choose “Password1” if required to include an uppercase letter and a number, or “Password1!” if a symbol is also required.
> 
> Users also express frustration when attempts to create complex passwords are rejected by online services.

johanstokking, 2020-02-16:
@jpmeijers this was the first check that returned an error. The [default policy](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/internal/shared/identityserver/config.go#L53-L56) is minimum length of 8 characters, 1 special char and 1 uppercase char, and it can be changed by anyone who deploys the stack.

This issue is about exposing that config to the front-end, so that we can have nice client-side validation, without hitting the back-end, and seeing this less-UX-friendly error.

bafonins, 2020-10-29:
I beleive it is already implemented. https://github.com/TheThingsNetwork/lorawan-stack/pull/3064 exposes all password related configuration and it is possible to configure those via `is.user-registration.password-requirements.{min_length|max_length|min_uppercase|min_special|min_digits}` flags.

@pgalic96 please reopen if you think that anything is missing.

closed 2020-10-29 by bafonins