TheThingsNetwork/lorawan-stack #1069: Update sigLen parameter in the CUPS HTTP response to 4 bytes

KrishnaIyer, 2019-07-30:
#### Summary

Update sigLen parameter in the CUPS HTTP response to 4 bytes.

#### Why do we need this?

This parameter has been updated in the [specification](https://doc.sm.tc/station/cupsproto.html#http-post-response).

#### What is already there? What do you see now?

sigLen is set to 2 bytes.

#### What is missing? What do you want to see?

sigLen should be 4 bytes.

#### Environment

`ttn-lw-stack v3.1.0`

#### How do you propose to implement this?

Simple change

#### Can you do this yourself and submit a Pull Request?

Yes

cc: @htdvisser

johanstokking, 2019-08-07:
Why backlog?

KrishnaIyer, 2019-08-07:
We don't support firmware updates yet.

htdvisser, 2019-08-07:
It's an easy, non-important fix, so anyone can pick this up when they're bored and looking for something to do.

PR #3192 (2020-09-10): Update sigLen parameter to 4 bytes

#### Summary

Closes #1069 

#### Changes

- Update `sigLen` parameters in both the marshaling and unmarshaling functions.
- Update test

#### Testing

UT

##### Regressions

None that I can think of

#### Notes for Reviewers

Not sure why this was bumped in priority and added to this milestone but done.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `READM [... cut]
closed 2020-09-10 by KrishnaIyer