TheThingsNetwork/lorawan-stack #1072: Wrong field mask for headers

adriansmares, 2019-07-30:
#### Summary

The field mask used for requests that provide `headers` is broken, in a sense that it attempts to specify the keys of the field as part of the field mask, even though they are not supposed to be specified. I think this would also apply to `attributes` (device, application etc.).

#### Steps to Reproduce

1. Create a Webhook creation request. I've used the following object:
```
{
  "ids": {
    "application_ids": {},
    "webhook_id": "wh1"
  },
  "base_url": "http://example.com/webhooks",
  "format": "json",
  "headers": {
    "k1": "v1",
    "k2": "v2"
  },
  "uplink_message": null,
  "join_accept": null,
  "downlink_ack": null,
  "downlink_nack": null,
  "downlink_sent": null,
  "downlink_failed": null,
  "downlink_queued": null,
  "location_solved": null
}
```
2. Attempt to `api.application.webhooks.update(appId, webhookId, patch)` (I've encountered this inside `ApplicationIntegrationAdd.handleSubmit`).

#### What do you see now?

The request will look as follows:
```
{
	"webhook": {
		"ids": {
			"application_ids": {},
			"webhook_id": "wh2"
		},
		"base_url": "http://example.com/webhooks",
		"format": "json",
		"headers": {
			"k1": "v1",
			"k2": "v2"
		},
		"uplink_message": null,
		"join_accept": null,
		"downlink_ack": null,
		"downlink_nack": null,
		"downlink_sent": null,
		"downlink_failed": null,
		"downlink_queued": null,
		"location_solved": null
	},
	"field_mask": {
		"paths": ["ids", "ids.application_ids", "ids.application_ids.application_id", "ids.webhook_id", "base_url", "format", "headers", "headers.k1", "headers.k2", "uplink_message", "join_accept", "downlink_ack", "downlink_nack", "downlink_sent", "downlink_failed", "downlink_queued", "location_solved"]
	}
}
```
The important part being `"headers", "headers.k1", "headers.k2", `.

#### What do you want to see instead?

Only the `headers` field should be specified.

#### Environment

196e6e66591b1a6482c7c3375f213b5de310fe59

#### How do you propose to implement this?

I think that fields such as `headers` and `attributes` shouldn't get parsed recursively when it comes to field masks. I think https://github.com/TheThingsNetwork/lorawan-stack/blob/196e6e66591b1a6482c7c3375f213b5de310fe59/sdk/js/src/util/marshaler.js#L113-L121 needs to skip these nodes. Thoughts on this @kschiffer ?

#### Can you do this yourself and submit a Pull Request?

Yes.

adriansmares, 2019-07-30:
Blocks #938

kschiffer, 2019-07-31:
I think hardcoding to skip these is not the way to go. What we need is a way to get the information of valid field masks from the backend and use it as a way to tell how to assemble field masks.

In case of the end device service, we already do this. We use the [`device-field-masks.json`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/sdk/js/generated/device-field-masks.json), from which we generate the [`device-entity-map.json`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/sdk/js/generated/device-entity-map.json). This info is used to [dertermine valid field masks for get/set](https://github.com/TheThingsNetwork/lorawan-stack/blob/f49a45148bd10eb600535d1b61e5617779a657c9/sdk/js/src/service/devices/index.js#L81-L99) operations.

For devices, the logic is a bit more complicated because we also need to determine which field mask paths correspond to which stack component `ns, js, is, as`. For other entites, this is not necessary.

So what we need is another field mask json for applications, consisting of an array of valid field masks for the application registry. We can use this array to check how field masks should be applied.

The logic for this is:
1. Check the path of the current prop
2. Cut off any bits from the path that extend after the valid matching field mask

The main problem with this approach is that the generation of the field mask json is not automated yet. I needed to copy it out myself from the [ttnpb package](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/ttnpb/field_mask_validation.go). The ones relevant for the application can be found via `applicationserver(_*)?.pb.paths.fm.go`. So based on how we will proceed with https://github.com/TheThingsNetwork/lorawan-stack/issues/32, automating this should be added later.

htdvisser, 2019-08-01:
We generate the API into JSON for a reason:

```js
"use strict";

const api = require("./sdk/js/generated/api.json");

const messages = {};

api.files.forEach(file => {
  if (!file.messages) return
  file.messages.forEach(message => { messages[message.fullName] = message })
})

const getMessageFields = (name, prefix = '') => {
  const message = messages[name]
  if (!message) return []
  var fields = []
  message.fields.forEach(field => {
    fields.push(prefix + field.name)
    if (field.label.includes('repeated')) return
    fields = fields.concat(getMessageFields(field.fullType, prefix + field.name + '.'))
  })
  return fields
}

getMessageFields('ttn.lorawan.v3.ApplicationWebhook')
```

kschiffer, 2019-08-01:
Ha! It never occurred to me that this info was implicitly sitting in the `api.json` as well. Way to go then.

PR #1087 (2019-08-02): Make allowed field masks part of the JS SDK

#### Summary

Closes #1072 
References #938

#### Changes

- Added a new mage target that exports `ttnpb.AllowedFieldMaskPathsForRPC` to a json file for the JS SDK.
- Changed the `api-definitions.json` generator to also include the allowed field masks next to the RPC methods.
- Added a way to filter field masks generated from object patches.

#### Notes for Reviewers

While I really liked @htdvisser solution from [the issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/1072#issuecomment-517163102), unfortunately it does not scale really  [... cut]
closed 2019-08-02 by adriansmares