TheThingsNetwork/lorawan-stack #1086: Add prop-types to all react components

bafonins, 2019-08-01:
#### Summary

A lot of react component in our codebase dont have prop-types.

#### Why do we need this?

Make sure that our components receive correct props, otherwise see errors in the dev tools console.
This will also help us to move to a stricter `eslint` config in the future.

#### What is already there? What do you see now?

A lot of component without full list of prop-types defined.

#### What is missing? What do you want to see?

All components having prop-types

#### Environment

browser (console,oauth)

#### How do you propose to implement this?

Simply define prop-types for each 

#### Can you do this yourself and submit a Pull Request?

yes. help is welcome

kschiffer, 2019-08-02:
Additionally, we need to decide on a way to add these prop-types. We currently mix
```js
static propTypes = { … }
// and
Component.propTypes = { … }
```

a lot. Static prop-types can have the advantage that they take props passed by HOCs into account. On the other hand, static prop-types only work for class components.
Another thing I noticed is that jsdoc comments of static prop-types will not be displayed in storybook, yet that might be fixable (though I wasn't able to fix it quickly last time I tried).

kschiffer, 2019-08-19:
So let's make this a little more actionable.
I propose the following rules:
- All react components, regardless of view, containers, plain components; regardless of functional or class **must have** prop-types and default-props defined
  - The only exception is components that don't take any props at all
  - The `children` prop does not need to be defined in prop-types / default-props
  - Any prop that is not set as required, needs to have a default-prop and vice-versa
  - We use `PropTypes.shape({ … })` to define expected object shapes in detail
  - We use our `webui/lib/prop-types.js` library to define prop-types and also add recurring type-definitions there
  - We *will* use an eslint config that notifies about missing prop-types
- **For functional components:**
  - We assign prop-types and default props definition as object property `MyFunctionalComponent.propTypes = { … }` / `MyFunctionalComponent.defaultProps = { … }`
  - We do not use default arguments for the functions as replacement for `defaultProps` definitions
- **For class components:**
  - We use the static property of the class `static propTypes = { … }` / `static defaultProps = { … }`
  - We do not use the constructor function to assign these (unless necessary)

Please thumb-up if you agree with these or write a comment about what you'd like to change.
The codebase does obviously not yet adhere to these rules, but we should make sure to work towards this successively.

Additionally, we need an issue to resolve the description bug in storybook for `static` pro-types (see my other comment).

bafonins, 2019-08-19:
> The children prop does not need to be defined in prop-types / default-props

**All** prop-types should be defined.

> We will use an eslint config that notifies about missing prop-types

1. Missing prop-types
2. Absence of default values for non-required props
3. Unused but defined props
4. Unsorted prop-types definitions
5. Usage of ambiguous types, i.e. `PropTypes.{array|object|any}` 

Ill create a PR adding these rules to show warnings only.

kschiffer, 2019-08-19:
Updated my comment. Alphabetical sorting is a bit too strict for my likings, but if you feel strongly about it, we can keep it.

johanstokking, 2019-11-12:
How far are we with this?

kschiffer, 2019-11-12:
We're currently doing this "as we go" and we already refactored quite some of the components, but still a lot to go. It would be good to do try refactoring the rest at once, though it would be quite some refactoring work.

@pgalic96 are you up to picking this up? You can check with `node_modules/.bin/eslint ./pkg/webui/**/*.js`.

pgalic96, 2019-11-14:
Yep, I can start revisiting all components

johanstokking, 2020-05-08:
What is the status of this one?

closed 2020-05-12 by bafonins