TheThingsNetwork/lorawan-stack #1094: Generate allowed field mask paths on webui build

bafonins, 2019-08-02:
#### Summary

`allowed-field-mask-paths.json` should be generated on `js:build` or `jsSDK:build` targets. Or `mage jsSDK:allowedFieldMaskPaths jsSdk:definitions` should be added to the docume tation.

#### Why do we need this?

The js sdk/webui build should take care of all build deps

#### What is already there? What do you see now?

`js:build` `jsSDK:build` targets

#### What is missing? What do you want to see?

`mage jsSDK:allowedFieldMaskPaths jsSdk:definitions` being run when building the sdk/webui

#### Can you do this yourself and submit a Pull Request?

i can, but will leave to the author of https://github.com/TheThingsNetwork/lorawan-stack/pull/1087

adriansmares, 2019-08-02:
https://github.com/TheThingsNetwork/lorawan-stack/blob/589b8193c321dd36ca9332b5206326b496540753/.mage/js_sdk.go#L143
https://github.com/TheThingsNetwork/lorawan-stack/blob/589b8193c321dd36ca9332b5206326b496540753/.mage/js_sdk.go#L63
Since allowed field mask paths is part of definitions, and definitions is part of build, isn't build already doing this ?

bafonins, 2019-08-02:
@adriansmares try on fresh repo

`make init`
`mage js:deps js:build`

PR #1098 (2019-08-05): dev: Generate allowed field mask paths before definitions

#### Summary

Closes #1094 

#### Changes

- Ensure that the `allowed-field-mask-paths.json` is generated before using it as a source in the `jsSDK:definitions` target.
- Make the generation smarter by using the corresponding ttnpb file as a source for `allowed-field-mask-paths.json`
closed 2019-08-05 by adriansmares