TheThingsNetwork/lorawan-stack #1121: Adding back deleted application not working

adamsondelacruz, 2019-08-07:
#### Summary
When you delete an application and decides to add it back, if fails because it complains that the ID already exists.

#### Steps to Reproduce

1. Remove existing application
2. Add it back using CLI or 
3. Add it back using API

#### What do you see now?
```
{
    "code": 6,
    "message": "error:pkg/identityserver/store:id_taken (ID already taken)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/identityserver/store",
            "name": "id_taken",
            "message_format": "ID already taken",
            "correlation_id": "e68a49d8b874408d97c8c57b15b0838e"
        }
    ]
}
```

#### What do you want to see instead?

Successfullly add back the application

#### Environment

amazon linux using TTN CLI and Postman using API Call

#### How do you propose to implement this?
When user try to insert a deleted application, update the deleted flag to false

#### Can you do this yourself and submit a Pull Request?

...

htdvisser, 2019-08-07:
Related to https://github.com/TheThingsNetwork/lorawan-stack/issues/1120

htdvisser, 2019-08-07:
We'd also need to check if 

1. the caller is a user or organization (not an application API key)
2. the caller was previously a member of the entity
3. the caller had sufficient rights on the entity (perhaps make RIGHT_APPLICATION_DELETE also restore)

When restoring the application, we should also consider that the deleted application may have had other members and API keys associated to it. The caller will need to indicate what to do with those, so this may actually require a dedicated API call.

htdvisser, 2019-08-07:
Turns out we already have another issue for this: https://github.com/TheThingsNetwork/lorawan-stack/issues/604

htdvisser, 2020-08-24:
Looks like I forgot to close this issue before. This is being tracked in #604.

closed 2020-08-24 by htdvisser

htdvisser, 2021-02-17:
This is implemented, and will land in `v3.12`.

tomascharad, 2021-05-29:
This is happening to me, but for a gateway

https://nam1.cloud.thethings.network/console/gateways/add

[image]

htdvisser, 2021-05-31:
This is an expected error if a regular user removed a gateway. 

The EUI is released, but the ID remains reserved. See also https://www.thethingsindustries.com/docs/reference/id-eui-constraints/#deleted-entities

Only when an admin user "purges" the gateway, the ID is released.

PR #3739 (2021-02-17): Soft-Deleted Entity Handling

#### Summary

This PR improves handling of soft-deleted entities in the Identity Server.

Refs #604 
Closes #1903 
Closes #1121

#### Changes

- API: Add `deleted_at` field to API of entities that can be soft-deleted
- API: Add ~`include_deleted`~ `deleted` to `List` and `Search` RPCs in registries with soft-delete
- API: Add `Restore` RPCs to registries with soft-delete
	- This requires the DELETE right, meaning that anyone who can delete the application, can also restore it.
- API: Add missing `Purge` RPC to OAuth client registry
- CONF: Add "restore window" con [... cut]